bblanchon/ArduinoJson #1063: deserialization issue

mars000, 2019-08-02:
Hi
I'm using latest library version 6.11.13
I don't know why I'm getting a crash when I deserializeJson.
The json string is correctly formatted...any help would be appreciated

this is the code snippet (little messy as I'm doing some basic tests but you will get idea of whats happening by tracking the Serial prints in the output below and I've also added the exception decoder output below...:
```cpp
 String message = "";

  while (LoRa.available())
  {
    message += (char)LoRa.read();
  }
  Serial.print(F("158 - message = "));
  Serial.println(message);
  Serial.println(F("160 -"));
  
  const int MAX_DOC_SIZE = 600;
    Serial.println(F("163 -"));
  
  int msg_length = message.length();
    Serial.println(F("166 -"));
  
  if (msg_length > MAX_DOC_SIZE)
  {
    Serial.println(F("328 - ERROR: JSON message recieved too large !"));
  }
  else
  {   
 Serial.println(F("174 -"));
  
    DynamicJsonDocument doc(1024); //  1024
    Serial.println(F("177 -"));
  
    DeserializationError error = deserializeJson(doc, message);
    Serial.println(F("180 -"));
    serializeJson(doc, Serial);
    Serial.println(F("182 -"));
    if (doc.containsKey("sync"))
    {
      Serial.println(F("185 - time sync received from gateway"));
#if defined(relay1)
      delay(1);
#elif defined(relay2)
      delay(500);
#elif defined(relay3)
      delay(1000);
#endif
    }
    else if (doc.containsKey("status"))
    {
      if (doc["status"] == "reset")
      {
        Serial.println(F("181 - status = restart received"));
        ESP.restart();
      }
    }

```

Serial Output = 
```cpp
LoRa Simple Node
Only receive messages from gateways
Tx: invertIQ disable
Rx: invertIQ enable

Send Message : 
Send Message : 
Send Message : 
Send Message : 
158 - message = {"sync":1300000}
160 -
163 -
166 -
174 -
177 -
Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
Core 1 register dump:
PC      : 0x400d10c1  PS      : 0x00060c31  A0      : 0x800d129e  A1      : 0x3ffbe470  
A2      : 0x3ffbe583  A3      : 0x00000001  A4      : 0x19999999  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x0013d620  A9      : 0x000000d0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0xffffffff  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000e  EXCCAUSE: 0x00000004  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
Core 1 was running in ISR context:
EPC1    : 0x400d10c1  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082a2d

Backtrace: 0x400d10c1:0x3ffbe470 0x400d129b:0x3ffbe560 0x400d13c2:0x3ffbe5e0 0x400d159d:0x3ffbe620 0x400d1391:0x3ffbe680 0x400d1406:0x3ffbe6c0 0x400d1ce7:0x3ffbe700 0x400d2317:0x3ffbe760 0x400d2322:0x3ffbe780 0x40080e91:0x3ffbe7a0 0x40081875:0x3ffbe7c0 0x400835c6:0x3ffb1ef0 0x400833c7:0x3ffb1f10 0x4008116b:0x3ffb1f30 0x400d0e77:0x3ffb1f50 0x400d1ba1:0x3ffb1f70 0x400d4b95:0x3ffb1fb0 0x40085095:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57
```

For what its worth the exception decoder output:
[image]

bblanchon, 2019-08-03:
Hi @mars000,

Despite the backtrace, this is not an error in the parser.
See this demo: https://wandbox.org/permlink/wwbTHmTv2R9ADGfp

I'm not familiar with the ESP exceptions, so I would look for:

* stack or heap corruption,
* problem with interruptions,
* hardware issue.

I don't think the error is in the snippet you posted.
Can you create an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)?

Best Regards,
Benoit

mars000, 2019-08-03:
thanks for response - I will try to product a MCVE.
I did get it working using MessagePack and I do have a somewhat unrelated question on that i.e. when I use the https://msgpack.org/   Try! tool I get a different compression % benefit from when I measure the length within the ArduinoJson code using measureMsgPack(doc)  -  i.e. measureMsgPack(doc) displays a smaller length than what the Try! tool produces; is that expected ?

bblanchon, 2019-08-11:
ArduinoJson uses the smallest type possible (for example fixmap instead of map32), so it's possible that the result is smaller than the demo.

Did you make some progress on your issue?

mars000, 2019-08-25:
hi - sorry for delay - been working on other projects but its all working well using
```
   DeserializationError error = deserializeMsgPack(doc, message);
```

closed 2019-08-25 by mars000