bblanchon/ArduinoJson #1064: Getting bytes in json and send with IR

ad1k4h, 2019-08-03:
Dear Everyone,
Im new to the platform and programming too, i would like to get IR Commands via JSON and send it with IRSEND. Here is my code:

```
server.onRequestBody([](AsyncWebServerRequest * request, uint8_t *data, size_t len, size_t index, size_t total) {
    Serial.println("Running");
    if (request->url() == "/test") {
      DynamicJsonBuffer jsonBuffer;
      JsonObject& root = jsonBuffer.parseObject((const char*)data);
      if (root.success()) {
        if (root.containsKey("command")) {
          Serial.println(root["command"].asString()); // Hello
          uint8_t cmd = root["command"];

          
          irsend.sendGree(cmd);
          
        }
      }
      request->send(200, "text/plain", "end");
    }
  });
```

I sending the code like this:
curl -i -X POST -H 'Content-Type: application/json' -d '{"command": "0x19, 0x00, 0x60, 0x50x00, 0x00, 0x00, 0x30"}' http://192.168.0.185/test

It is arriving, but not seem to working.

I have a working code, but the ir commands are hardcoded. Example:

//Declaration
uint8_t Power_ON[8]  = {0x19, 0x09, 0x60, 0x50, 0x00, 0x00, 0x00, 0xC0};

//Called like
irsend.sendGree(Power_ON);

Im have this issue a week ago, and no one else can help me :( Thank you!

ad1k4h, 2019-08-03:
Update: I Tried to send with this method, but not working:
curl -i -X POST -H 'Content-Type: application/json' -d '{"command": "[0x19, 0x00, 0x60, 0x50x00, 0x00, 0x00, 0x30]"}' http://192.168.0.185/test

bblanchon, 2019-08-05:
Hi @ad1k4h,

I don't think JSON is a good fit for your project, because JSON (and ArduinoJson) doesn't support hexadecimal.

If I were you, I would not use JSON (nor any other encoding), and I would send the raw binary data in the request's body.

Best Regards,
Benoit

barbiani, 2019-08-05:
Why can not he send the stringified json bytes?

bblanchon, 2019-08-07:
"Why not use a string?" is this what you mean?
I think it defeats the purpose of using JSON in the first place, and you'd have to parse the string in the microcontroller.

gigs2go, 2019-08-14:
The declaration of Power_ON makes it an 8-byte array - which is what 'irsend.sendGree(Power_ON);' does.
You will have to map your request (however it is formatted) into the appropriate 8-byte structure, and send that.

ad1k4h, 2019-08-14:
Thank you for all the answers, i dont know how to do that. Im new to this.
Can u please help me for maing this work ? Thank you!

bblanchon, 2019-08-22:
@ad1k4h, I'm sorry but ArduinoJson keeps me busy enough, I cannot afford to help users with unrelated topics.
Thank you for your understanding.

closed 2019-08-22 by bblanchon