bblanchon/ArduinoJson #1065: ArduinoJson not reading all the way

mailmartinviljoen, 2019-08-05:
When I read json data is seems to stop at a sertain object.

Below is my json string
```

{
"blockregister":
 [
			{
				"node_name":"ESP00002",
				"shared_key":"123456",
				"serv_count":"554",
				"slot_1":"test",
				"slot_2":"slot2"
			},
			{
				"node_name":"ESP00003",
				"shared_key":"123456",
				"serv_count":"341",
				"slot_1":"notest",
				"slot_2":"slot2"
			}
]
}
```

I retrieve the objects like this and all variables are populated successfully.
``
  const char* node_name = doc["blockregister"][0]["node_name"];
  const char* shared_key = doc["blockregister"][0]["shared_key"];
  const char* serv_count = doc["blockregister"][0]["serv_count"];
  const char* slot_1 = doc["blockregister"][0]["slot_1"];
  const char* slot_2 = doc["blockregister"][0]["slot_2"];
```
If I try read the second set of objects like this.
``
  const char* node_name = doc["blockregister"][1]["node_name"];
  const char* shared_key = doc["blockregister"][1]["shared_key"];
  const char* serv_count = doc["blockregister"][1]["serv_count"];
  const char* slot_1 = doc["blockregister"][1]["slot_1"];
  const char* slot_2 = doc["blockregister"][1]["slot_2"];
```
It will only read up to slot1,  const char* slot_2  is populated with nothing.
Are there any limitations or am i doing something obviously wrong ?

bblanchon, 2019-08-07:
Hi @mailmartinviljoen,

Check the return value of [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/), it probably says `NoMemory`, which means the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) is too small.

Indeed, to improve the performance, reduce code size, and prevent heap fragmentation, ArduinoJson uses a fixed-allocation strategy. The drawback is that you must create a [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) large enough before-hand.

It can be tricky to compute the right capacity for the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/), but the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) is here to help you.

Best Regards,
Benoit

mailmartinviljoen, 2019-08-09:
Thanks I figured it out. I was using the first example I could from the examples which was working with chars, Without really investigating. 

The HTTP client returns the payload into a String. Which I converted with Payload.toCharArray 
After using some more. after doing this as you said

`  DynamicJsonDocument doc(19000);
  deserializeJson(doc, Payload);
  JsonObject obj = doc.as<JsonObject>();`

Solved my problem. I  guess I could set the capacity to the payload length.

Thanks again.

bblanchon, 2019-08-10:
Be careful; the capacity of the `JsonDocument` is not the same thing as the Payload length.
The `JsonDocument` stores a tree of `JsonArray`, `JsonObject`, and `JsonVariant`; the size is usually larger than the JSON string.

closed 2019-08-10 by bblanchon