bblanchon/ArduinoJson #1068: NotSuported (Timestamp). Difference Wandbox to µC

Chris685, 2019-08-09:
Hello,

why is my JSON working on Wandbox with the "timestamp" but not working at the µC. 

With Wandbox i am using  "char json[] =", on the µC i use the WiFi Client. The Timestamp information is not needed for me.

bblanchon, 2019-08-10:
Hi @Chris685,

I'm sorry, I don't understand your question.
Can you share the code snippet?

Best Regards,
Benoit

Chris685, 2019-08-10:
Sorry,

i will write more details - my suggestion is, that with the definition of  const char json[] it works different.

- Code from Wallbox (is working)

`// ArduinoJson - arduinojson.org
// Copyright Benoit Blanchon 2014-2019
// MIT License
//
// This example shows how to deserialize a JSON document with ArduinoJson.

#include <iostream>
#include "ArduinoJson.h"

int main() {
  // Allocate the JSON document
  //
  // Inside the brackets, 200 is the capacity of the memory pool in bytes.
  // Don't forget to change this value to match your JSON document.
  // Use arduinojson.org/v6/assistant to compute the capacity.
  //StaticJsonDocument<6694> doc;

  // StaticJsonDocument<N> allocates memory on the stack, it can be
  // replaced by DynamicJsonDocument which allocates in the heap.
  //
   const size_t capacity = 3*JSON_ARRAY_SIZE(1) + 2*JSON_ARRAY_SIZE(25) + JSON_OBJECT_SIZE(2) + 27*JSON_OBJECT_SIZE(3) + 26*JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(11) + 2990;
    DynamicJsonDocument doc(capacity);

  // JSON input string.
  //
  // Using a char[], as shown here, enables the "zero-copy" mode. This mode uses
  // the minimal amount of memory because the JsonDocument stores pointers to
  // the input buffer.
  // If you use another type of input, ArduinoJson must copy the strings from
  // the input to the JsonDocument, so you need to increase the capacity of the
  // JsonDocument.
  const char json[] =
      "{\"language\":\"de\",\"headers\":[{\"id\":\"00_0010_7751_0100\",\"description\":\"VBus 0: DeDietrich Diemasol C v2007\",\"channel\":0,\"destination_address\":16,\"source_address\":30545,\"protocol_version\":16,\"command\":256,\"info\":0,\"destination_name\":\"DFA\",\"source_name\":\"DeDietrich Diemasol C v2007\",\"fields\":[{\"id\":\"000_2_0\",\"name\":\"Temperatur Sensor 1\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"002_2_0\",\"name\":\"Temperatur Sensor 2\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"004_2_0\",\"name\":\"Temperatur Sensor 3\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"006_2_0\",\"name\":\"Temperatur Sensor 4\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"008_2_0\",\"name\":\"Temperatur Sensor 5\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"010_2_0\",\"name\":\"Temperatur Sensor 6\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"012_2_0\",\"name\":\"Temperatur Sensor 7\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"014_2_0\",\"name\":\"Temperatur Sensor 8\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"016_2_0\",\"name\":\"Temperatur Sensor 9\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"018_2_0\",\"name\":\"Temperatur Sensor 10\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"020_2_0\",\"name\":\"Temperatur Sensor 11\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"022_2_0\",\"name\":\"Volumenstrom\",\"unit\":\" l/min\",\"unit_code\":\"LitersPerMinute\"},{\"id\":\"024_1_0\",\"name\":\"Drehzahl R1\",\"unit\":\"%\",\"unit_code\":\"Percent\"},{\"id\":\"025_1_0\",\"name\":\"Drehzahl R2\",\"unit\":\"%\",\"unit_code\":\"Percent\"},{\"id\":\"026_1_0\",\"name\":\"Drehzahl R3\",\"unit\":\"%\",\"unit_code\":\"Percent\"},{\"id\":\"027_1_1\",\"name\":\"Relaisstatus R4\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"027_1_2\",\"name\":\"Relaisstatus R5\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"027_1_4\",\"name\":\"Relaisstatus R6\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"027_1_8\",\"name\":\"Relaisstatus R7\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"027_1_16\",\"name\":\"Relaisstatus R8\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"027_1_32\",\"name\":\"Relaisstatus R9\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"028_4_0\",\"name\":\"Wärmemenge\",\"unit\":\" Wh\",\"unit_code\":\"WattHours\"},{\"id\":\"032_4_0\",\"name\":\"Systemdatum\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"036_2_0\",\"name\":\"Systemzeit\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"039_1_0\",\"name\":\"Variante\",\"unit\":\"\",\"unit_code\":\"None\"}]}],\"headerset_stats\":{\"headerset_count\":1,\"min_timestamp\":1565245304.893,\"max_timestamp\":1565245304.893},\"headersets\":[{\"timestamp\":1565245304.893,\"packets\":[{\"header_index\":0,\"timestamp\":1565245304.674,\"field_values\":[{\"field_index\":0,\"raw_value\":30,\"value\":\"30.0\"},{\"field_index\":1,\"raw_value\":27.4,\"value\":\"27.4\"},{\"field_index\":2,\"raw_value\":40.4,\"value\":\"40.4\"},{\"field_index\":3,\"raw_value\":50.4,\"value\":\"50.4\"},{\"field_index\":4,\"raw_value\":46.2,\"value\":\"46.2\"},{\"field_index\":5,\"raw_value\":888.8,\"value\":\"888.8\"},{\"field_index\":6,\"raw_value\":888.8,\"value\":\"888.8\"},{\"field_index\":7,\"raw_value\":24.2,\"value\":\"24.2\"},{\"field_index\":8,\"raw_value\":888.8,\"value\":\"888.8\"},{\"field_index\":9,\"raw_value\":888.8,\"value\":\"888.8\"},{\"field_index\":10,\"raw_value\":888.8,\"value\":\"888.8\"},{\"field_index\":11,\"raw_value\":0,\"value\":\"0.0\"},{\"field_index\":12,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":13,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":14,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":15,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":16,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":17,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":18,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":19,\"raw_value\":0,\"value\":\"0\"},{\"field_in [... cut ...]
        p_kollektor.setPic(8);          // Dach mit Sonne und Rauch
      }
      if ((Pumpe_Fest == 0) && (Pumpe_prim == 0)) {           // Bild Standart
        p_feuer.setPic(6);              // Kamin
        p_kollektor.setPic(4);          // Dach
      }
    }
    else
    {
      Serial.println("Fehler in response");
    }
    http.end();
#ifdef DEBUG_JSON
    Serial.println("JSON vBUS abgeschlossen - Verbindung beendet");
#endif
  }
  else {
#ifdef DEBUG_JSON
    Serial.println("WLAN nicht verfügbar");
#endif
  }
}
`

bblanchon, 2019-08-10:
This is not really the same code, is it? :-)

Please see:

* [Why does ArduinoJson return `NotSupported`?](https://arduinojson.org/v6/faq/why-does-arduinojson-return-notsupported/)

Chris685, 2019-08-10:
OK, i will show this behaviour with the same code:

The difference:

Once with http.getString() (not supported):
	  DeserializationError error = deserializeJson(doc, http.getString());
Once with json const char (working):
	  //DeserializationError error = deserializeJson(doc, json);

The char is converted from your assistent on your homepage.

```
     const char json[] = "{\"language\":\"de\",\"headers\":[{\"id\":\"00_0010_7751_0100\",\"description\":\"VBus 0: DeDietrich Diemasol C v2007\",\"channel\":0,\"destination_address\":16,\"source_address\":30545,\"protocol_version\":16,\"command\":256,\"info\":0,\"destination_name\":\"DFA\",\"source_name\":\"DeDietrich Diemasol C v2007\",\"fields\":[{\"id\":\"000_2_0\",\"name\":\"Temperatur Sensor 1\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"002_2_0\",\"name\":\"Temperatur Sensor 2\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"004_2_0\",\"name\":\"Temperatur Sensor 3\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"006_2_0\",\"name\":\"Temperatur Sensor 4\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"008_2_0\",\"name\":\"Temperatur Sensor 5\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"010_2_0\",\"name\":\"Temperatur Sensor 6\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"012_2_0\",\"name\":\"Temperatur Sensor 7\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"014_2_0\",\"name\":\"Temperatur Sensor 8\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"016_2_0\",\"name\":\"Temperatur Sensor 9\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"018_2_0\",\"name\":\"Temperatur Sensor 10\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"020_2_0\",\"name\":\"Temperatur Sensor 11\",\"unit\":\" °C\",\"unit_code\":\"DegreesCelsius\"},{\"id\":\"022_2_0\",\"name\":\"Volumenstrom\",\"unit\":\" l/min\",\"unit_code\":\"LitersPerMinute\"},{\"id\":\"024_1_0\",\"name\":\"Drehzahl R1\",\"unit\":\"%\",\"unit_code\":\"Percent\"},{\"id\":\"025_1_0\",\"name\":\"Drehzahl R2\",\"unit\":\"%\",\"unit_code\":\"Percent\"},{\"id\":\"026_1_0\",\"name\":\"Drehzahl R3\",\"unit\":\"%\",\"unit_code\":\"Percent\"},{\"id\":\"027_1_1\",\"name\":\"Relaisstatus R4\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"027_1_2\",\"name\":\"Relaisstatus R5\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"027_1_4\",\"name\":\"Relaisstatus R6\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"027_1_8\",\"name\":\"Relaisstatus R7\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"027_1_16\",\"name\":\"Relaisstatus R8\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"027_1_32\",\"name\":\"Relaisstatus R9\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"028_4_0\",\"name\":\"Wärmemenge\",\"unit\":\" Wh\",\"unit_code\":\"WattHours\"},{\"id\":\"032_4_0\",\"name\":\"Systemdatum\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"036_2_0\",\"name\":\"Systemzeit\",\"unit\":\"\",\"unit_code\":\"None\"},{\"id\":\"039_1_0\",\"name\":\"Variante\",\"unit\":\"\",\"unit_code\":\"None\"}]}],\"headerset_stats\":{\"headerset_count\":1,\"min_timestamp\":1565245304.893,\"max_timestamp\":1565245304.893},\"headersets\":[{\"timestamp\":1565245304.893,\"packets\":[{\"header_index\":0,\"timestamp\":1565245304.674,\"field_values\":[{\"field_index\":0,\"raw_value\":30,\"value\":\"30.0\"},{\"field_index\":1,\"raw_value\":27.4,\"value\":\"27.4\"},{\"field_index\":2,\"raw_value\":40.4,\"value\":\"40.4\"},{\"field_index\":3,\"raw_value\":50.4,\"value\":\"50.4\"},{\"field_index\":4,\"raw_value\":46.2,\"value\":\"46.2\"},{\"field_index\":5,\"raw_value\":888.8,\"value\":\"888.8\"},{\"field_index\":6,\"raw_value\":888.8,\"value\":\"888.8\"},{\"field_index\":7,\"raw_value\":24.2,\"value\":\"24.2\"},{\"field_index\":8,\"raw_value\":888.8,\"value\":\"888.8\"},{\"field_index\":9,\"raw_value\":888.8,\"value\":\"888.8\"},{\"field_index\":10,\"raw_value\":888.8,\"value\":\"888.8\"},{\"field_index\":11,\"raw_value\":0,\"value\":\"0.0\"},{\"field_index\":12,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":13,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":14,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":15,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":16,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":17,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":18,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":19,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":20,\"raw_value\":0,\"value\":\"0\"},{\"field_index\":21,\"raw_value\":4662206,\"value\":\"4662206\"},{\"field_index\":22,\"raw_value\":20190808,\"value\":\"20190808\"},{\"field_index\":23,\"raw_value\":502,\"value\":\"8:22\"},{\"field_index\":24,\"raw_value\":1,\"value\":\"1\"}]}]}]}";  

void readJSON() {
#ifdef DEBUG_JSON
  Serial.println(F("JSON vBUS starten"));
#endif
  // Verbinde zu Server
  if (WiFi.status() == WL_CONNECTED) {
    HTTPClient http;  //Object of class HTTPClient
    http.begin("http://192.168.1.25/dlx/download/live?sessionAuthUsername=admin&sessionAuthPassword=XXX");
    http.begin(dl2_adresse);
    int httpCode = http.GET();
    //String payload = http.getString();
#ifdef DEBUG_JSON
    Serial.print("Response Code: "); //200 is OK
    Serial.println(httpCode);
    //Serial.print("Returned data from Server:");
    //Serial.println(payload);    //Print request response payload
#endif

    if (httpCode == 200) {
      // Parsing
#ifdef DEBUG_JSON
      Serial.println("Syntaxanalyse JSON vBUS");
#endif
[... cut ...]
      const char* vBusSpeicher_mitte = doc["headersets"][0]["packets"][0]["field_values"][4]["value"];
      const char* vBusSpeicher_unten = doc["headersets"][0]["packets"][0]["field_values"][2]["value"];
      const char* vBusFestbrennstoff = doc["headersets"][0]["packets"][0]["field_values"][7]["value"];
      const char* vBusPumpe_primaer = doc["headersets"][0]["packets"][0]["field_values"][12]["value"];
      const char* vBusPumpe_Festbrennstoff = doc["headersets"][0]["packets"][0]["field_values"][15]["value"];
...
```

bblanchon, 2019-08-10:
Did you try [ARDUINOJSON_DECODE_UNICODE](https://arduinojson.org/v6/api/config/decode_unicode/) as suggested?

Chris685, 2019-08-10:
Hi,

unluckly the same Result. Still not supported.

I put 

```c++
#define ARDUINOJSON_DECODE_UNICODE 1
```
and

```c++
      StaticJsonDocument<11000> doc;
```
into the code, because with dynamic the code was not longer compiled

bblanchon, 2019-08-10:
The only situation where `deserializeJson()` returns `NotSupported` is when the input contains a Unicode escape sequence and `ARDUINOJSON_DECODE_UNICODE` is 0.

Make sure to define `ARDUINOJSON_DECODE_UNICODE` before including `ArduinoJson.h`, as shown in the documentation.

If it still doesn't work, please provide an [MVCE](https://stackoverflow.com/help/minimal-reproducible-example).

BTW, you should avoid `http.getString()` and instead use `http.getStream()` to reduce memory usage. You can find a complete example in the GitHub case study in [Mastering ArduinoJson](https://arduinojson.org/book/), but there are also many examples on the Internet.

Chris685, 2019-08-10:
I am sorry - and you are right.

The Problem was the define of `ARDUINOJSON_DECODE_UNICODE` after `ArduinoJson.h`.

Now i have upgraded my project from Arduino DUE to ESP8266 and all newest Librarys.

Thanks a lot for your time.

closed 2019-08-10 by Chris685

bblanchon, 2019-08-11:
You're welcome @Chris685. 
Thank you for using ArduinoJson.