bblanchon/ArduinoJson #1072: Fails to compile in PlatformIO due to long path

manuelbl, 2019-08-11:
The library fails to compile in PlatformIO with the below error message:
```
.pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/../Variant/../Polyfills/type_traits.hpp:9:45: fatal error: type_traits/integral_constant.hpp: No such file or directory
```
The problem only occurs on Windows and only if the PlatformIO project is in a location with long path name.

In my case, the project location `C:\Users\SomeUser\Documents\PlatformIO\Projects\Nano-ArduinoJson` worked while the path `C:\Users\SomeUser\Documents\LongPathLongPathLongPath\LongPathLongPath\LongPathLongPath\Nano-ArduinoJson` failed.

Note that PlatformIO downloads the library to a directory within the project directory and compiles it there. So the full include path to the include file becomes:

```
C:\Users\SomeUser\Documents\PlatformIO\Projects\Nano-ArduinoJson\.pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/../Variant/../Polyfills/type_traits.hpp
```

Even though a good part of the path length is caused by PlatformIO, it would seem that the ArduinoJson part `ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/../Variant/../Polyfills` has ample opportunity for improvement. At the same time, the cyclomatic complexity of the software would be reduced.

Workaround: Move the project to a location with a shorter path.

To reproduce use the below minimal PlatformIO project:

**platformio.ini**

```
[env:nanatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
lib_deps =
    ArduinoJson
```

**main.cpp**

```
#include <Arduino.h>
#include "ArduinoJson.h"

void setup() {
}

void loop() {
}
```

**Build log**

```
> Executing task: C:\Users\SomeUser\.platformio\penv\Scripts\platformio.exe run <

Processing nanoatmega328 (platform: atmelavr; board: nanoatmega328; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/nanoatmega328.html
PLATFORM: Atmel AVR 1.15.0 > Arduino Nano ATmega328
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 30KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.1.1   
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...    
Dependency Graph        
|-- <ArduinoJson> 6.11.3
Compiling .pio\build\nanoatmega328\src\main.cpp.o
Archiving .pio\build\nanoatmega328\libfad\libArduinoJson_ID64.a
Archiving .pio\build\nanoatmega328\libFrameworkArduinoVariant.a
Compiling .pio\build\nanoatmega328\FrameworkArduino\CDC.cpp.o
Compiling .pio\build\nanoatmega328\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\nanoatmega328\FrameworkArduino\HardwareSerial0.cpp.o
Compiling .pio\build\nanoatmega328\FrameworkArduino\HardwareSerial1.cpp.o
Compiling .pio\build\nanoatmega328\FrameworkArduino\HardwareSerial2.cpp.o
In file included from .pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/../Variant/VariantSlot.hpp:8:0,
                 from .pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/MemoryPool.hpp:9,
                 from .pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson/Array/../Variant/../Misc/../Strings/StringAdapters.hpp:7,
                 from .pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson/Array/../Variant/../Misc/SerializedValue.hpp:7,
                 from .pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson/Array/../Variant/VariantData.hpp:7,
                 from .pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson/Array/ArrayRef.hpp:7,
                 from .pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson.hpp:17,
                 from .pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson.h:9,
                 from src\main.cpp:2:
.pio\libdeps\nanoatmega328\ArduinoJson_ID64\src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/../Variant/../Polyfills/type_traits.hpp:9:45: fatal error: type_traits/integral_constant.hpp: No such file or directory
compilation terminated.
C*ompiling .pio\build\nanoatmega328\FrameworkArduino\HardwareSerial3.cpp.o
** [.pio\build\nanoatmega328\src\main.cpp.o] Error 1
================================================================================== [ERROR] Took 1.65 seconds ==================================================================================
The terminal process terminated with exit code: 1
```

bblanchon, 2019-08-12:
Hi @manuelbl,

Thank you for this detailed report.

I initially used relative include path to simplify the usage of the library, and I didn't see this problem coming.
I don't see a way to fix this while preserving the relative includes, so I'll have to switch to paths relative to the root of the library.

Since it's a risky change, I'll have to wait for 6.12.
In the meantime, please use the [single file distribution](https://github.com/bblanchon/ArduinoJson/releases/download/v6.11.3/ArduinoJson-v6.11.3.h)

Best Regards,
Benoit

closed 2019-08-26 by bblanchon

bblanchon, 2019-09-05:
Fix available in [ArduinoJson 6.12.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.12.0).

commit fcbec6eb (2019-08-26): Use absolute instead of relative includes (fixes #1072)