bblanchon/ArduinoJson #1073: Fails to compile on Particle.io WebIDE

shanevanj, 2019-08-13:
Fails to compile on Particle.io WebIDE

Error on WebIDE is 
`lib/ArduinoJson/ArduinoJson/Strings/SizedFlashStringAdapter.hpp:19:69: 'strncmp_P' was not declared in this scope`

Particle.io WebIDE target device Gen3 Mesh (Argon or Xenon)
Library version ArduinoJson (6.11.4)    

Reproduce
Open WebIDE
Select libraries, Search ArduinoJSON, select JsonParserExample.ino then select `USE THIS EXAMPLE`
Compile the App and error is produced

bblanchon, 2019-08-13:
Hi @shanevanj,

Thank you for reporting this.

It seems to be a regression introduced by #1071.
I guess that Particle defines `PROGMEM` but doesn't implement the `strncmp_P()`.
I wish we had a Continuous Integration for Particle, but I tried and it seemed impossible.

As a workaround, you can define [`ARDUINOJSON_ENABLE_PROGMEM`](https://arduinojson.org/v6/api/config/enable_progmem/) to `0`, or roll back to 6.11.3.

Best Regards,
Benoit

bblanchon, 2019-08-13:
Indeed, only `strcmp_P()` is defined 🙄 :
https://github.com/particle-iot/device-os/blob/1b1805fb4afcb93b0e92fbecfd28c082d6310285/wiring/inc/spark_wiring_arduino.h

shanevanj, 2019-08-14:
> you can define ARDUINOJSON_ENABLE_PROGMEM to 0, or roll back to 6.11.3.

Thanks I tried both 

> #include "Arduino.h"
> #include <Particle.h>
> 
> #define ARDUINOJSON_ENABLE_PROGMEM 0

and 

> Included libraries
> 
> ArduinoJson (6.11.3)    
> LiquidCrystal_I2C_Spark (1.1.0)    
> MQTT (0.4.29)    
> PollingTimer (1.3.0)    
> PublishQueueAsyncRK (0.0.5)
> 

 and the compile still fails with same error - I guess its because the entry is missing in https://github.com/particle-iot/device-os/blob/1b1805fb4afcb93b0e92fbecfd28c082d6310285/wiring/inc/spark_wiring_arduino.h 

I edited the same file on my local workbench install ( added in the strncmp_P() ) and the code  compiles locally okay. 

I have raised a bug request on particle (https://github.com/particle-iot/device-os/issues/1884) which they are now arguing about.

If I insert this the .h file above where the <ArduinoJson.h> is first used it compiles 

> 
> #ifndef strncpy_P
> #define strncpy_P strncpy
> #endif
> 
> #ifndef strncmp_P
> #define strncmp_P strncmp
> #endif
> 
> #define ARDUINOJSON_ENABLE_PROGMEM 0

bblanchon, 2019-08-14:
Hi @shanevanj,

Thanks for the update.

You must define `ARDUINOJSON_ENABLE_PROGMEM` before including `ArduinoJson.h` (be careful because the Web IDE insert `#include` statements at the top of the file); I checked, and it works.

You're right about 6.11.3: it doesn't work either.
That's good news because it means it's not a regression from #1071 

Thanks for opening a bug request on Particle.
I'll continue to work on a workaround on my side: I'll try to detect when the `PROGMEM` macro is empty.

Best Regards,
Benoit

shanevanj, 2019-08-14:
Thanks for your patience ! - issue closed on particle with comment that 

> <Arduinio.h> 

> #define PARTICLE_WIRING_ARDUINO_COMPATIBILTY 1

must be included before any PROGMEM dependant code. 

Regards Shane

closed 2019-08-21 by bblanchon

bblanchon, 2019-08-23:
A fix is included in [ArduinoJson 6.11.5](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.11.5)

commit f00dfd7b (2019-08-21): Added fallbacks strlen_P, strncmp_P, strcmp_P and memcpy_P (fixes #1073)