bblanchon/ArduinoJson #1075: Saving JSON to file  question

mistergreen, 2019-08-20:
This is more general question. I'm planning to save multiple device settings onto a SD card, JSON format, for recovery situations. When I make a change on 1 device, can I save and affect that one device JSON branch in the SD card? Or do I need to gather all device settings, and rewrite all of the JSON data on the SD card?

bblanchon, 2019-08-22:
Hi @mistergreen,

I'm sorry I don't understand the question.
Can you clarify?

Best Regards,
Benoit

mistergreen, 2019-08-22:
I'll try to be more clear. Can I alter and save a node (in a JSON tree) on the SD card or is it easier to create a whole new JSON tree with the altered node and unaltered nodes and save that to the SD card?

bblanchon, 2019-08-22:
No, ArduinoJson doesn't allow editing a value in a file.
Instead, you must load the whole file, edit the value, and save the whole file.

closed 2019-08-22 by mistergreen