bblanchon/ArduinoJson #1076: Library can't parse JSON with [] as parent

yuriyfedyuk, 2019-08-21:
Hello,
I'm using ESP32 board.

Can't parse JSON with next format
[{
"product_id":"0000"
}]

This JSON can be parsed:
{
"product_id":"0000"
}

Web assistant shows that it should be parsed.

bblanchon, 2019-08-22:
Hi @yuriyfedyuk,

Did you call [parseArray()](https://arduinojson.org/v5/api/jsonbuffer/parsearray/) or [parseObject()](https://arduinojson.org/v5/api/jsonbuffer/parseobject/)?
Are you sure the [JsonBuffer](https://arduinojson.org/v5/api/jsonbuffer/) is big enough?

Best Regards,
Benoit

yuriyfedyuk, 2019-08-22:
Hi @bblanchon,

I'm sure that JsonBuffer is big enough. As an example, I used a small part of my JSON string.
In code I call parseObject().

yuriyfedyuk, 2019-08-22:
It works with parseArray() call.

closed 2019-08-22 by yuriyfedyuk