bblanchon/ArduinoJson #1078: Cant get a data from Json string V6

GdLakeIoT, 2019-08-24:
Hi,
I have been tying to get all that data from the string below.
{"nodeId":2755343285,"subs":[{"nodeId":3290279261,"root":true,"subs":[{"nodeId":2013506667}]}]}

I have replaced the " so the string is 

"{\"nodeId\":2755343285,\"subs\":[{\"nodeId\":3290279261,\"root\":true,\"subs\":[{\"nodeId\":2013506667}]}]}"

What I need exactly need it to be able to get the NodeID that has Root = True
I just cant seem to get it to work.
Any help, links or suggestion will be appreciated.

bblanchon, 2019-08-28:
Hi @GdLakeIoT,

Please check out the [examples](https://arduinojson.org/v6/example/) and the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/).

Best Regards,
Benoit

bblanchon, 2019-08-29:
You're welcome George!
Don't hesitate to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support ArduinoJson ;-)
Good luck with your project.