bblanchon/ArduinoJson #1079: How to deserialize json in arduino JSON 6

Project786, 2019-08-26:


bblanchon, 2019-08-28:
Hi @Project786,

What is the problem exactly?

Best Regards,
Benoit

Project786, 2019-09-02:
Thankyou so much for your reply.

The scenario is as follows:

--I am trying to send data from Arduino Mega to Nodemcu using the Arduino
Json library. I am connecting the TX of the Nodemcu to the RX of the
Arduino Mega and the RX of the Nodemcu to the TX of the Arduino Mega.
--I am able to send data using Arduino Json 5 library.
--This is the link I used to send data using Arduino Json 5 library:*
https://mybtechprojects.tech/serial-communication-between-nodemcu-and-arduino/
<https://mybtechprojects.tech/serial-communication-between-nodemcu-and-arduino/>.*

My issue is as follows:

--I am able to send data with Arduino Json 5 library so there is no problem
with that.
--*My main issue is that I want to use the newer Arduino Json library,
which is version 6, but I'm not able to have success with that.*

I have attached four *.ino* files for your reference. The first two files
are using the Arduino Json 5 library which works nicely.
In the other two files, I've tried using Arduino Json 6 but without success.

I have named the files appropriately.

I hope this email finds you well and I look forward to hearing from you
soon.

Thanks
Project786

bblanchon, 2019-09-02:
>  but I'm not able to have success with that

What do you mean?
Did you get a compilation error? a runtime error? a crash?

Project786, 2019-09-02:
No errors, no crashes.

Its just that after uploading, I open the serial monitor of the Nodemcu
side, I get null and 0 values.
With Json 5 I receive the data at the Nodemcu side but not with Json 6.

ewaldc, 2019-09-02:
When you say "no error", does _deserializeJson(doc, Serial);_ return no error ?
I am having a challenge with deserialize from stream too, but in my case I am getting an error code back (but no crash)

Project786, 2019-09-02:
I don't get any errors with any statement. The only thing is with the serial monitor which isn't showing any value.

bblanchon, 2019-09-05:
We don't have enough information to help you.
Can you share both the v5 and v6 programs? I may be able to spot the mistake.

closed 2019-09-25 by bblanchon