bblanchon/ArduinoJson #1082: deserializeJson() raises NoMemory when reading from a (SPIFFS) stream

ewaldc, 2019-08-31:
I recently ported my code from the wonderful v5 library to v6.  There was superb porting documentation and thus 95% of porting was done quickly.   The most challenging part was losing a truly DynamicJson object/buffer (see #1074), but as I was cleaning up code that cached all JSON source strings in memory (as the JsonDocument now does that for you), I hit the following issue. 

On ESP8266 platform (PlatformIO IDE), when trying to deserialize a JSON string from a stream (SPIFFS file), things go well when reading the stream into a char* using a single readBytes call.
When reading directly from the stream, it fails with _NoMemory_.   The json string is 2480 bytes, and deserialization of the string succeeds _after_ failure to deserialize from file, so it does not seem to be a platform (heap) space issue.  

I read issue #862 but in that case the problem was different.

Things verified:
- Their is sufficient space reserved (doubling or tripling JsonDoc size makes no difference)
- The file contains a single line compressed JSON string (no spaces, no tabs, no CR/LF, no \0)
- Adding a \0 does not help, in fact there is also no \0 in the char array for which parsing succeeds as I omitted declaring an extra byte and adding \0

Things not done:
- Adding an empty line (CR/LF)
- Reading from the file after the error occurs to see where it stopped
- Parsing another JSON string
- Have not looked at the code...

Happy to share the JSON string if needed.

```
File f = dir.openFile("r");
if (f) {
 int sz = f.size();
 DynamicJsonDocument jsonDoc1(sz), jsonDoc2(sz);
 jsonStr = new char [sz];
 f.readBytes(jsonStr, sz);
 f.seek(0); //reset file pointer to beginning
DeserializationError error1, error2;
error1 = deserializeJson(jsonDoc1, f, DeserializationOption::NestingLimit(20)); //fails
error2 = deserializeJson(jsonDoc2, jsonStr, DeserializationOption::NestingLimit(20)); //succeeds
}
```
Any idea why this would run out of memory?  It might be a stack issue if somehow the code  would read into a 2480 byte local variable as ESP has 4K max stack.

* target platform : ESP8266 (nodemcu 1.0)
* compiler model and version: PlatformIO Core 4.0.3

bblanchon, 2019-09-02:
Hi @ewaldc,

I see a small misunderstanding. You use the size of the JSON input to set the capacity of the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/), but that's not enough: ArduinoJson needs a bit more memory to store the JSON object tree.

Use the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) to compute the capacity, and don't forget to include the "additional bytes for strings duplication", since you read directly from a string.

Why did this code work with ArduinoJson 5? Because [`DynamicJsonBuffer`](https://arduinojson.org/v5/api/jsonbuffer/) could grow when needed. I removed this feature because it had too many drawbacks.

Best regards,
Benoit

ewaldc, 2019-09-02:
Hello Benoit,
Thanks for taking the time !!!

Of course I understand that ArduinoJson may require more space than the size of the JSON string.
In the full code this is part of a loop which increases the size of the _JsonDocument_ 
until deserialize succeeds. I have explained that in my comment on issue (see #1074) as an illustration of why v6 code is more complex in certain cases such as in my code.

The JSON string is 2539 bytes and the _JsonDocument_ requires _only_ 2480 bytes.
Of course, before posting the issue, I additionally used the Assistant to validate that the size is big enough. 

In v6, for user supplied JSON strings (unknown at compile time), one has to make a guess of the initial size to use for the _JsonDocument_. It just happens to be that the size of the source Json string is a very good starting value.  For my code, I evaluated around 520 strings and _string.length()_ was right as initial choice for around ~95% of json strings while not wasting more than 10% space in 80% of the time.  Just for small JSON strings with lots of integers, this is too small but adding 256 bytes was enough in all but one exception.   

The issue which I filed is that using the **same** sized _DynamicJsonDocument_  (of sufficient size) the deserialization succeeds when I read the file first into a string/char array, but fails when I deserialize from a stream.
This could never work if the space would be insufficient, unless somehow the same JSON string would require more space when read from file.  But that would be highly unlikely, is it not?

Ewald

bblanchon, 2019-09-02:
Hi Ewald,

[`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) behaves differently when it reads from a read-only input, like a file.
For read-only inputs, it requires more space because it makes a copy of each string.
For `char*` inputs, it requires less space because it mutates the input and stores pointers.
Please see the documentation.

Best Regards,
Benoit

ewaldc, 2019-09-03:
Ah, got it.  Still learning the implementation details of V6...
So similar to V5, one has to keep the original Json string in memory (or as a class member) if it is NOT read-only, a const or a stream.  And, if it is read-only, increase the size of the _DynamicJsonDocument_  with the size of the source Json string to allow duplication.   I wrongly assumed _JsonDocument_ holds all the data, but it just holds the JsonBuffer.
It's working now.

closed 2019-09-03 by ewaldc

bblanchon, 2019-09-03:
If you want to make a copy of a read-write input (i.e., a `char*`), simply cast it to `const char*`:

```c++
deserializeJson(doc, (const char*)input);
```

Now, the `JsonDocument` holds everything.

ewaldc, 2019-09-09:
Wonderful tip, this provides full flexibility.  

On the ESP platform though, I generally prefer not to duplicate the (writeable/constructed) source string because it requires twice the memory available and has caused my code to run out of memory for larger JSON strings. Speed has never been a problem though, so the extra copy cycles do not matter.
That is how I came to use ArduinoJson in the first place.  No other implementation was/is so memory efficient!
Maybe if I write the JSON strings to flash/SPIFFS first... 
Would definitely reduce complexity of class structures and scope/(de)allocation challenges related with managing the source string alongside the _JsonDocument_.