bblanchon/ArduinoJson #1083: Seeking to check for nested containskey()

lbussy, 2019-09-01:
Similar to #322, I am seeking to find if a key which may be nested exists.  However, I don't think #322 works for me.  Please allow me to explain the application:

JSON may appear similar to the following:

```
{
    "hostname": "foo",
    "itemconfig": {
        "name": "bar",
        "thing": false
    },
    "target1config": {
        "targeturl": "http://something.local/target.php",
        "freq": "60"
    },
    "target2config": {
        "targeturl": "http://somethingelse.local/target.php",
        "freq": "60"
    }
}
```

As you can see, I have ["target1config"]["freq"] and ["target2config"]["freq"].  An endpoint may receive a POST containing partial JSON as a configuration change.  I need to see if the key is present in the JSON to determine if I should update my configuration. 

JsonObject::containsKey() seems to do what I want however you have stated that this will not work with nested keys.  #322 will check for the presence of a nested key but as you can see I have the same nested key name in a different portion of the JSON.

The logic I am attempting to facillitate is as follows:

```
if keyExists(["target1config"]["freq"])
    strlcpy(config->freq1, doc["target1config"]["freq"], sizeof(config->freq1));
```

If this is what #322 is supposed to allow, I suppose I am not understanding it.

bblanchon, 2019-09-02:
Hi @lbussy,

Here is the way to do it with 6.11:

```c++
if (!["target1config"]["freq"].isNull())
    strlcpy(config->freq1, doc["target1config"]["freq"], sizeof(config->freq1));
```

or better:

```c++
const char* freq1 = doc["target1config"]["freq"];
if (freq)
    strlcpy(config->freq1, freq1, sizeof(config->freq1));
```

Alternatively, if you want to use a default value:

```c++
strlcpy(config->freq1, doc["target1config"]["freq"] | "default", sizeof(config->freq1));
```

Best Regards,
Benoit

ewaldc, 2019-09-02:
Hi @lbussy 
I am using a recursive function.  This example recursively looks for a key called _link_, extracts the URL that corresponds with it and registers a callback handler to process it.   The variable _obj_ contains the "parent", which you can use to test the context e.g. to match ["light"]["link"] but not ["temp"]["link"].  If you want to make it generic, you can pass the key as an argument rather than to hard code it as "link" as in my example.  
```
void OpenHab::registerLinkHandlers(const JsonVariant prototype) {
	if (prototype.is<JsonObject>()) {
		const JsonObject& obj = prototype;
		for (const JsonPair pair : obj) {
			if (!strcmp(pair.key().c_str(), "link")) {
				String link = pair.value().as<char *>();
          			_server.on(link, std::bind(&OpenHab::handleLink, this, (JsonVariant) obj));
			}
			registerLinkHandlers(pair.value().as<JsonObject>());
			for (auto arr : pair.value().as<JsonArray>()) {
				registerLinkHandlers(arr.as<JsonVariant>());
			}
		}
	} else if (prototype.is<JsonArray>()) {
		const JsonArray& arr = prototype;
		for (const auto& elem : arr) registerLinkHandlers(elem.as<JsonVariant>()); //, parent);
	}
}
```
PS. this code is for v6, I have a v5 alternative...
Hope this helps

lbussy, 2019-09-02:
Thank you Benoît and Ewald.  I believe this gets me exactly what I need.

lbussy, 2019-09-02:
Actually, this failed on one of my use cases.

```
bool dospiffs = doc["firstlevel"]["doIt"];
if (doIt)
    config->doIt= doIt;
```

NULL == false in this case so I can't ignore a NULL condition, it will always set it to false.  Seems like with C++ NULL is defined as 0 or 0L so I'm not quite sure what to do with this part of it, although your suggestion worked fgreat on the other keys.

lbussy, 2019-09-02:
For my purposes, I was able to use .indexOf() to check if the key existed, however that feels less than sexy.

bblanchon, 2019-09-02:
Here is how to fix the `bool` use-case:

```c++
bool dospiffs = doc["firstlevel"]["doIt"];
if (doIt)
    config->doIt= doIt;

lbussy, 2019-09-03:
I thought I tried both of those - maybe not.   Holiday here today so I have been out.  Will try tomorrow morning.  Thank you.

lbussy, 2019-09-03:
That works, thank you Benoît.  Not sure what I tried previously - I guess it wasn't that. :)

bblanchon, 2019-09-04:
You're welcome @lbussy. Thank you for using ArduinoJson.

closed 2019-09-04 by bblanchon