bblanchon/ArduinoJson #1085: Mechanize nested brackets

polclota, 2019-09-03:
I haven't been able to find a way to mechanize the switch part of this code. Any suggestion?

`String b = "MQTT/server/broker";
String c = "Firebase/active";
String d = "timestamp";

String printJ(String r)
{
  if (r.equals(""))
    return "";
  byte i = count_chars(r.c_str(), '/') + 1;
  String a[i];
  for (byte x = 0; x < i; x++)
  {
    a[x] = r.substring(0, r.indexOf('/'));
    r = r.substring(r.indexOf('/') + 1);
  }
  switch (i)
  {
  case 1:
    return jsonConfigBuffer[a[0]].as<String>();
    break;
  case 2:
    return jsonConfigBuffer[a[0]][a[1]].as<String>();
    break;
  case 3:
    return jsonConfigBuffer[a[0]][a[1]][a[2]].as<String>();
    break;
  case 4:
    return jsonConfigBuffer[a[0]][a[1]][a[2]][a[3]].as<String>();
    break;

  default:
    break;
  }
}
void setup()
{
  Serial.println(printJ(b));
  Serial.println(printJ(c));
  Serial.println(printJ(d));
}`

bblanchon, 2019-09-03:
Hi @polclota,

This is very similar to the JSON Path feature requested in #801.
Please upvote it, if that's something you'd be interested in.

Meanwhile, I think you can simplify the loop and get rid of the switch. 
Something along those lines:

```c++
JsonVariant result = jsonConfigBuffer;
for (byte x = 0; x < i; x++)
  {
    result = result[r.substring(0, r.indexOf('/'))];
    r = r.substring(r.indexOf('/') + 1);
  }
return result;
```

Best Regards,
Benoit

polclota, 2019-09-03:
Many thanks, it worked for me but I had to add: ` .as<JsonVariant>()` to the definition.

`  byte i = count_chars(r.c_str(), '/') + 1;
  JsonVariant result = jsonConfigBuffer.as<JsonVariant>();
  for (byte x = 0; x < i; x++)
  {
    result = result[r.substring(0, r.indexOf('/'))];
    r = r.substring(r.indexOf('/') + 1);
  }
  return result;`

bblanchon, 2019-09-04:
If the value exists in the document and you only want to replace it, keep this code and call [`JsonVariant::set()`](https://arduinojson.org/v6/api/jsonvariant/set/).

Creating a value would be more complicated: you'd have to call [`createNestedObject()`](https://arduinojson.org/v6/api/jsonobject/createnestedobject/) in the loop.

polclota, 2019-09-04:
Great, that's it, here it goes!

`JsonVariant result = jsonConfigBuffer.as<JsonVariant>();
  for (byte x = 0; x < i; x++)
  {
    result = result[r.substring(0, r.indexOf('/'))];
    r = r.substring(r.indexOf('/') + 1);
  }
  result.set(v) ;`

bblanchon, 2019-09-05:
Cool. I guess we can close this issue then?

polclota, 2019-09-05:
Sure and thanks a lot.

bblanchon, 2019-09-05:
You're welcome @polclota.
Don't forget to star the project to show your support 😉

closed 2019-09-05 by bblanchon