bblanchon/ArduinoJson #1086: Deserialize GeoJson data with double precision number

EmmanuelMiquet, 2019-09-05:
Hi,
I am trying to use ArduinoJson to parse some GeoJson data (geodetic positions of landmarks from a map database).
However, I am facing an issue, as floating point precision (C++ float32) is not enough to not loose precision in the map (lines sampling points not aligned anymore etc...).
I checked that ARDUINOJSON_USE_DOUBLE is set to 1 in my msvc solution.
I tried to use "as<double>()" to read, and the values have indeed lots of decimals, but seem to loose precision. I also checked using "as<std::string>()" and casting it to double, and could see that the returned string had only float32 precision (6 right digits).
Am I missing anything? Or is this a known/normal behavior?
Thanks in advance,
Emmanuel

bblanchon, 2019-09-06:
Hi Emmanuel,

I'm afraid [ARDUINOJSON_USE_DOUBLE](https://arduinojson.org/v6/api/config/use_double/) is your only option. There are always precision issues with floating points values, but I'm surprised that the accuracy isn't good enough for your application.

`as<std::string>()` will not work because the value was already converted to a `float` by `deserializeJson()`. To avoid this, you can transmit a string in the JSON document instead of a float (ie add quotes around the values).

Best Regards,
Benoit

EmmanuelMiquet, 2019-09-10:
Hi, sorry for the late answer, 
Thanks for the details. Indeed, storing coordinates as strings in the JSON seems to be the simplest approach.
Best regards,
Emmanuel

closed 2019-09-10 by EmmanuelMiquet