bblanchon/ArduinoJson #1087: mqtt uint16_t port from struct

ElizabethMorves, 2019-09-08:
I ask for help. This is not a library issue. This is a very wonderful thing and I do not know anything better. I use a standard example JsonConfigFile.ino but outside the main code. So....
class adafruit_mqtt_client need this
` public:
  Adafruit_MQTT_Client(Client *client, const char *server, uint16_t port,
                       const char *cid, const char *user, const char *pass):
    Adafruit_MQTT(server, port, cid, user, pass),
    client(client)
  {}`
all fields come all fields come except uint16_t port.
well, maybe the problem is Adafruit?

`Adafruit_MQTT_Client mqtt(&client,
                            config.mqtt_server,
                            config.mqtt_port | 1883,
                            config.mqtt_cid,
                            config.mqtt_username,
                            config.mqtt_password
                          );`
**config.mqtt_port | 1883**  choose your own. But this does not end my troubles.
librarys refuses to work NTP, NeoPixel  and the rest. Some kind of street magic happens with
uint16_t but char works fine.
My struct Config wery simple

`
struct Config{
uint16_t
          mqtt_port,
          ntp_timeoffset;
char
          mqtt_server[64],
          mqtt_username[64],
          mqtt_password[64],
          ntp_server[128];
};`

I spent a lot of time trying to find a problem, but couldn’t localize it.
I appeal to you as an older friend for help. My knowledge is not good enough. 

Thanks for the long time

bblanchon, 2019-09-09:
Hi @ElizabethMorves,

ArduinoJson's [operator|](https://arduinojson.org/v6/api/jsonvariant/or/) only works on [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/).

The code above calls this operator on `uint16_t`; in that case, that's a binary OR operation.
The result of this operation is probably not what you expect.

You can fix this bug by either:

1. making sure that `config.mqtt_port` is not zero before calling `Adafruit_MQTT_Client` constructor, or
2. replacing with `config.mqtt_port != 0 ? config.mqtt_port : 1883` (which we can write `config.mqtt_port ?: 1883`, but it's a GCC specific extension)

Best Regards,
Benoit

ElizabethMorves, 2019-09-09:
Thanks for the quick response. I made this recording from hopelessness.
The main problem of my code, it's all types (uintx, int) are 0. At the same time, the type char works fine. I'm at a loss, how could this happen? After initializing the configuration, I call to the console 
`Serial.println(config.mqtt_port);`
and he is there, 1883  then i call the constructor and inside class...variable portnum suddenly... becomes equal 0. What happens to type char and doesn't happen to type int?
I don’t understand how to ask Google. I will go a difficult way. I'll start from the beginning. When I figure it out,  will describe it in the sky, in huge letters. I have enough strength and determination to start with a clean main(). This is the only way to get closer to your knowledge, master.

Thanks for the long time

closed 2019-09-10 by ElizabethMorves