bblanchon/ArduinoJson #1090: Second item in array not gets assigned from struct object

AnumSheraz, 2019-09-16:
Hi, 

First off, excellent library and thank you for your great efforts. 
I got this code from [this amazing](https://arduinojson.org/v6/assistant/) tool. 

```
const size_t capacity = JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(4) + 2*JSON_OBJECT_SIZE(7);
DynamicJsonDocument doc(capacity);

JsonObject config = doc.createNestedObject("config");
config["type"] = "power_plug";
config["chip_id"] = "12345";
config["mac_address"] = "abc";

JsonArray config_attachments = config.createNestedArray("attachments");

JsonObject config_attachments_0 = config_attachments.createNestedObject();
config_attachments_0["name"] = "socket";
config_attachments_0["input_type"] = "toggle_switch";
config_attachments_0["on_state"] = "on";
config_attachments_0["off_state"] = "off";
config_attachments_0["pin_type"] = "output";
config_attachments_0["pin"] = 15;
config_attachments_0["current_state"] = "on";

JsonObject config_attachments_1 = config_attachments.createNestedObject();
config_attachments_1["name"] = "socket_1";
config_attachments_1["input_type"] = "toggle_switch";
config_attachments_1["on_state"] = "on";
config_attachments_1["off_state"] = "off";
config_attachments_1["pin_type"] = "output";
config_attachments_1["pin"] = 12;
config_attachments_1["current_state"] = "off";

serializeJson(doc, Serial);
```
This prints out the desired json properly. but when ament single line from the second array item, to take the `on_state` from my struct object `d_config`, 
```
Serial.println(">>" + d_configs.peripherals[1].on_state);
config_attachments_1["on_state"] = d_configs.peripherals[1].on_state;
```
the parsing gets quiet unexpected, and some of the keys from serialised string json gets missing. For above case, This gets printed in serial monitor;

```
>>on
{"config":{"type":"power_plug","chip_id":"12345","mac_address":"abc","attachments":[{"name":"socket","input_type":"toggle_switch","on_state":"on","off_state":"off","pin_type":"output","pin":15,"current_state":"on"},{"name":"socket_1","input_type":"toggle_switch","on_state":"on","off_state":"off","pin_type":"output","pin":12}]}}
```
**NOTE** in the second item of `attachments`, the key `current_state` is entirely missing. 
When I assign all values from this struct object, the serialised json misses many more keys... and the behaviour is quite unexpected. here is how I have defined my structs;

```
typedef struct Peripheral{
    String name;
    String input_type;
    String on_state;
    String off_state;
    String pin_type;
    String current_state;
    int pin;
};

typedef struct {
  String type;
  String chip_id;
  String mac_address;
  Peripheral peripherals[10];    
} DeviceConfig;
```
I am not sure if its an issue during serialization or something else... :/ 
Any help would be highly appreciated :) .

AnumSheraz, 2019-09-16:
When I increase the capacity size, the serialization happens peropery
 const size_t capacity = JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(4) + 4*JSON_OBJECT_SIZE(10);

**NOTE** increased only the last one from `3*JSON_OBJECT_SIZE(7)` to `4*JSON_OBJECT_SIZE(10)`;
Seems like an issue with https://arduinojson.org/v6/assistant/ ?

noobGB, 2019-09-16:
Hi, 
I too facing somewhat similar kind of issue while migrating from version 5.13.2 to version 6.12.0 , below  is my simple json input,

{"led1StateJson":0,
"led2StateJson":0,
"led3StateJson":0,
"blynk_tokenJson":"some_value",
"server_idJson":"2",
"Wifi_SSID_Json":"some_value",
"Wifi_Pass_Json":"some_value"}

I have used JSON_OBJECT_SIZE(7) as per ArduinoJson assistant but last 3 variables are not getting serialised, if I use JSON_OBJECT_SIZE(11) then all are getting serialised.
It is strange because ArduinoJson version 5.13.2 works fine with JSON_OBJECT_SIZE(7).

AnumSheraz, 2019-09-16:
@noobGB, the Author replied here;
https://arduino.stackexchange.com/a/68659/13754

noobGB, 2019-09-16:
@AnumSheraz , Thank you !!

closed 2019-09-16 by AnumSheraz