bblanchon/ArduinoJson #1091: Feature request: periodic yield() in deserialization and/or error handling

PaintYourDragon, 2019-09-18:
Target system is an Adafruit M4 board using their TinyUSB library (to expose onboard flash as mass storage over USB), but it's possible ESP8266 could encounter a similar category of trouble due to its unrelenting watchdog timer. Arduino IDE 1.8.8.

Problem: if JSON deserialization fails (e.g. due to invalid syntax), error handling takes sufficiently long that TinyUSB (which requires periodic servicing) is knocked offline (and may in turn take Serial with it). This makes it impossible to access the JSON file to deduce the problem.

Temporary workaround: a delay(100) before deserializing improves the probability (but does not guarantee) that TinyUSB servicing takes place before whatever operation is taking a while...the mass storage filesystem stays online (usually) and one can review the problem JSON file.

More robust solution: a periodic call to yield() ...perhaps there's an outer loop in the deserialization, or something invoked per-line... and/or similar in the error handling case (which is when the problem most seems to manifest). A few times per second would likely suffice. Note that yield() is considered canon in the Arduino environment now, and on most platforms will define or compile to nothing, so it shouldn't be a space or performance concern on those platforms that don't require it.

Thanks!

CC @ladyada @hathach

bblanchon, 2019-09-19:
Hi @PaintYourDragon,

This looks like a feature for [StreamUtils](https://github.com/bblanchon/ArduinoStreamUtils).
We could add a new input stream decorator that call `yield()` every 10ms.
Would that solve your problem?

Best Regards,
Benoit

PaintYourDragon, 2019-09-19:
Howdy, thank you for the reply.
I've been able to resolve this in my case by slipping in some yield()s in the code immediately before and after deserialization. So it wasn't really the deserialization alone that lead to the problem, but this _plus_ time spent in my own code that pushed things just over the timeout threshold.
Gonna close the issue, since the library really isn't the sole culprit. But maybe keep the periodic yield()s in mind if you get reports from ESP8266 or TinyUSB users in the future.
Thanks!

closed 2019-09-19 by PaintYourDragon