bblanchon/ArduinoJson #1093: serializeJson output larger than measureJson() - caused by character encoding ?

ewaldc, 2019-09-19:
Occasionally the byte count from serializeJson() is larger than what is reported by, _measureJson()_, causing the web server to transmit partial JSON strings.
E.g. when requesting one of these JSON strings through curl, I receive the error:
**curl: (18) transfer closed with 3 bytes remaining to read**

So I instrumented the buffered print code to compare the size reported by _measureJson()_ with the actual bytes written to the stream and found that it happens once and awhile.  Below a snippet of how the JSON string is submitted using buffered print of 256 byte blocks (due to memory pressure).
```
void OpenHab::SendJson(JsonVariant obj) {
	DbgPrintln(F("OpenHab::SendJson "), measureJson(obj));	
	_server.setContentLength(measureJson(obj));
	_server.send(200, getContentTypeStr(APP_JSON));
	_print.client(_server.client());
	serializeJson(obj, _print);
	_print.stop();
}
```
[here](https://pastebin.com/MbAnSTLC) is a JSON string that is 3 bytes longer when serialized than reported by _measureJson()_. _measureJson()_ reflects the correct amount of bytes of the original source string received from the client (the one in the link)
The strings are uploaded by clients, so I have no control over them.

It looks like serializeJson encodes certain characters in unicode, but nowhere in de code I have _ARDUINOJSON_DECODE_UNICODE_ set

> original: "Temperature [22.0 °C]"
> serialized: "Temperature [22.0 ┬░C]"

* ArduinoJson 6.12
* target platform: ESP8266
* compiler model and version" platformio core 4.0.3

Thanks in advance,
Ewald

bblanchon, 2019-09-20:
Hi Ewald,

This is very surprising because `measureJson()` and `serializeJson()` share most of their code.

Can you check without the buffered print?
Can you replace the buffered print with this "official" [`WriteBufferingClient`](https://github.com/bblanchon/ArduinoStreamUtils)?
Can you provide an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)?

Best regards,
Benoit

ewaldc, 2019-09-21:
It's indeed very strange....
Never seen this issue with ArduinoJson 5.  It started as I ported to v6, but of course I also moved to the latest ESP8266 code (2.5.2 and master tested).  Probably a bug somewhere in my code, but I can't seem to find it... 

I can't print the whole string in any way due to insufficient memory, but I can print an object from the parsed string.  From the whole string I printed the  _"widgetId": "0202"_ object.
I simplified the code to not using the buffered print and add a Serial.println to reduce the chance that it's somewhere is ESP library:
```
ICACHE_FLASH_ATTR void SendJson(JsonVariant obj) {
	int size = measureJson(obj);
	char *buffer = new char[size+1];
	_server.setContentLength(size);
	DbgPrintln(F("free heap memory: "), ESP.getFreeHeap());
	serializeJson(obj, buffer, size);
	DbgPrintln(F("free heap memory: "), ESP.getFreeHeap());
	buffer[size] = '\0';
	Serial.println(buffer);
	_server.send(200, getContentTypeStr(APP_JSON), buffer);
	DbgPrintln(F("free heap memory: "), ESP.getFreeHeap());
	delete buffer;
}
```
It comes one byte short (the final **}**) per this JSON [**string**](https://pastebin.com/m7CLhVXs) both in _Serial.println_ as well as in __server.send()_
Per the documentation, I don't think  _serializeJson_ needs an extra byte for'\0', unless I am mistaken.
I have added the _ESP.getFreeHeap()_  checks to be sure it never runs out of space.
Will test also with my bufferedPrint to see if it's also one byte (had the impression it came two bytes short there...)

Thanks in advance,
Ewald

bblanchon, 2019-09-21:
Hi @ewaldc,

`serializeJson()` expects the size of the buffer as the third argument, so it should be:

```c++
serializeJson(obj, buffer, size+1);
```

You don't have to write the terminator, `serializeJson()` does it for you.

Best Regards,
Benoit

ewaldc, 2019-09-21:
Hi @bblanchon ,
For the webserver we don't want the extra '\0',  as it takes a char buffer as argument and the exact amount of bytes (thus not a "string" char array).  Compared to the true code, I have only added one extra byte to be able to do the Serial.println as well in the same function which of course requires a null terminated string/char* and thus requires one extra byte in the buffer.
As I understand it:

-  _measureJson()_ says how many bytes are needed in a buffer e.g. to send out from a webserver. Hence it does not include the '\0'
- from the API guide on serialize, I understand it writes to a buffer when one adds the size argument and not a string (so no '\0' at the end)

> 1. - the JsonDocument to serialize (in fact, you can also use a JsonArray, a JsonObject, or a JsonVariant)
> 2. - output: the destination where the JSON document should be written
> 3. - outputSize: the capacity of the destination **buffer**

Hence I was expected the buffer to contain _measureJson()_ bytes which represent the json "string" but without the '\0' so I can send the buffer to a webserver.

Here is a [**MCVE** ](https://pastebin.com/wvgMnNzc).  It's without webserver.  If needed I'll add the webserver part. 

Warm regards,
Ewald

ewaldc, 2019-09-22:
Made good progress this morning by capturing the byte count from _serializeJson(obj, buffer, size+1)_
Found so far:
-  _serializeJson_ does not really write to a buffer (aka char[]), but rather a null terminated char* (aka "string"), so it needs one more byte in size than what _measureJson()_  reports.  In other words, it writes (size - 1) bytes and sets buffer[size-1] = '\0').  
-  _serializeJson_  returns the same number of bytes as _measureJson()_, even though it writes one extra byte ('\0').
- when writing to Serial or a char* string, the bytes written by _serializeJson_ equals the byte size in an editor with code set UTF-8 (3966 in y demos/0202 example) as well as the size reported by _measureJson()_.  However when writing to my buffered print code it returns 4 characters less, yet the client displays the outcome perfectly.  There are 4 higher end UTF-8 characters involved, so there is probably a relationship.

IMHO it would make sense to provide some extra info e.g. in the serialization guide, maybe not use the term _buffer_  (but null terminated char* string) and making it clear you need to pass size+1 for the extra '\0').  

With that out of the way, the problem has now become: why did the ArduinoJson PrintWriter write 4 bytes less?  When looking at the source code, I noticed my buffered print code does not override _size_t write(const uint8_t* s, size_t n)_  Maybe the problem is there...

ewaldc, 2019-09-22:
Problem solved. The error was fully on my side: my buffered Print Writer class was not compatible with ArduinoJson 6.
For those interested, find [**here**](https://pastebin.com/eq4Qqyn9) an updated, simple buffered Print writer which writes to a WifiClient/ESP8266WebServer using a (small) buffer of your choice (e.g. 256 bytes).  This class allows to send a large (up to ~40K) json object on the ESP8266 platform without running out of memory.  Using a normal string and ESP webserver send you would need >3x the size of your object.
Use as follows:
```
WiFiClientPrint<256> _print;  // declare once, 256 byte buffer

void SendJson(JsonVariant obj) {
	int size = measureJson(obj);
	_server.setContentLength(size);
	_server.send(200, "application/json");
	_print.client(_server.client());  // _server is ESP8266WebServer, update to current client
	serializeJson(obj, _print);
	_print.stop();
}
```

closed 2019-09-22 by ewaldc

bblanchon, 2019-09-23:
Hi @ewaldc,

Thanks for the update.
Again, I strongly recommend using the "official" buffered print [WriteBufferingStream](https://github.com/bblanchon/ArduinoStreamUtils#buffering-write-operations):

```c++
WriteBufferingStream bufferedWifiClient{_server.client(), 256};
serializeJson(doc, bufferedWifiClient);
bufferedWifiClient.flush();
```

Best regards,
Benoit

ewaldc, 2019-09-25:
Will give it another try.  Tried a couple of months ago and got compilation errors on ESP platform, so I quickly wrote my own.   This time I will file an issue the error is still there....