bblanchon/ArduinoJson #1094: Unable to use ArduinoJson in two-file class structures

Xamtastic, 2019-09-22:
Hi, what a marvellous library you have ... but alas, I have encountered a problem!
# Introduction to the problem
I've created a class that uses ArduinoJson and when compiled on the Mac using a Makefile, it works fine, but when added to the setup of an Arduino sketch, it fails miserably. 

I created a small example to reproduce the problem and have put the extensive error notices at the bottom of this note. 

Unless its already available, what is needed is a two-file class structured Hello World example, maybe you can adapt the files below to that!

Meanwhile, how would I resolve these errors?

My aim is to have a C++ two-file class (i.e. `.h` and `.cpp`) that uses ArduinoJson but is agnostic to the platform that it is on - as you can see here it works properly when run compiled by `g++` on a Mac but the same platform agnostic class does not work when run by the Arduino IDE.

This example of the problem is really a cut-down of a class that I have written that has JSON snippets passed to it from files that are stored, and read from the ESP32's SPIFFS file system.

The reason why I want my ArduinoJson in a separate class like this is so that it can easily be unit tested on my build server, so what I really need to know is how to write C++ classes that can be built on a build server using g++ to prove that the code hasn't been broken by another developer, whilst also being able to be included in an Arduino sketch as a `.h` file include. 

# How I reproduced the Problem
My reproduction of the problem consists of the following five files:
1. Main.cpp (to run it on the desktop using the Makefile, which employs g++)
2. MyProcessor.h (the header file of the class that encapsulates ArduinoJson)
3. MyProcessor.cpp (the class file that encapsulates ArduinoJson functionality)
4. Makefile (requires make already installed, and the path to Arduino src changed in the setting at the top of the file)
5. ArduinoJsonHelloClassStructure.ino (the Arduino sketch that runs successfully without the above classes but generates the said errors that are at the bottom of this note)

# ArduinoJsonHelloClassStructure.ino

Run this file 'as-is' and it works perfectly.

Uncomment the code around `MyProcessor` (which encapsulates ArduinoJson in a two-file class structure, MyProcessor.h and MyProcessor.cpp), and it generates the errors at the foot of the page. 

```
#include <ArduinoJson.h>
#include "MyProcessor.h"
#include <string>

void setup() {
  Serial.begin(9600);
  
  const size_t capacity = JSON_OBJECT_SIZE(1) + 30;
  DynamicJsonDocument doc(capacity); 
  const char* json = "{\"ToPrintOut\":\"Hello World\"}"; 
  deserializeJson(doc, json); 
  const char* toPrintOut1 = doc["ToPrintOut"]; // "Hello World" 
  Serial.println(toPrintOut1);

  /*
  MyProcessor processor; 
  processor.loadJsonString(std::string(json)); 
  const char* toPrintOut2 = processor.getValue("ToPrintOut").c_str(); 
  Serial.println("Key: ToPrintOut, Value: " + String(toPrintOut2));
  */
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
# MyProcessor (MyProcessor.h and MyProcessor.cpp)
The code that works in the Arduino sketch is embedded into a two-file class, `MyProcessor.h` and `MyProcessor.cpp`.

## MyProcessor.h
Please note the private variable called `DynamicJsonDocument doc`
```
#ifndef MYPROCESSOR_H
#define MYPROCESSOR_H

#include <iostream>
#include <ArduinoJson.h>

class MyProcessor
{
[... 129 lines cut ...]
/Users/[redacted]/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:176:7: note: candidate: template<class TChar> typename ArduinoJson6120_000001::enable_if<ArduinoJson6120_000001::IsString<TChar*>::value, ArduinoJson6120_000001::VariantConstRef>::type ArduinoJson6120_000001::JsonDocument::operator[](TChar*) const
       operator[](TChar* key) const {
       ^
/Users/[redacted]/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:176:7: note:   template argument deduction/substitution failed:
sketch/MyProcessor.cpp:15:16: note:   mismatched types 'TChar*' and 'std::__cxx11::basic_string<char>'
  return doc[key];
                ^
In file included from /Users/[redacted]/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/BasicJsonDocument.hpp:7:0,
                 from /Users/[redacted]/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from /Users/[redacted]/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:21,
                 from /Users/[redacted]/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from sketch/MyProcessor.h:5,
                 from sketch/MyProcessor.cpp:1:
/Users/[redacted]/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:180:44: note: candidate: ArduinoJson6120_000001::ElementProxy<ArduinoJson6120_000001::JsonDocument&> ArduinoJson6120_000001::JsonDocument::operator[](size_t)
   FORCE_INLINE ElementProxy<JsonDocument&> operator[](size_t index) {
                                            ^
/Users/[redacted]/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:180:44: note:   no known conversion for argument 1 from 'std::__cxx11::string {aka std::__cxx11::basic_string<char>}' to 'size_t {aka unsigned int}'
/Users/[redacted]/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:184:32: note: candidate: ArduinoJson6120_000001::VariantConstRef ArduinoJson6120_000001::JsonDocument::operator[](size_t) const
   FORCE_INLINE VariantConstRef operator[](size_t index) const {
                                ^
/Users/[redacted]/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:184:32: note:   no known conversion for argument 1 from 'std::__cxx11::string {aka std::__cxx11::basic_string<char>}' to 'size_t {aka unsigned int}'
exit status 1
Error compiling for board ESP32 Dev Module.

```

ewaldc, 2019-09-22:
@Xamtastic ,
Can't say why your code does not compile (seems OK to me) but here is how I solve the issue with sizing the DynamicJsonDoc.  Basically, the first time I use a very large buffer (relative to free heap space) and once the exact size is known, I deserialize a final time but now with the exact space.  No stack fragmentation occurs, since its a large block and all freed right after allocation.
I know it looks a little clumsy, but I have not found a better way to allocate exactly the right size for a given JSON string.  ArduinoJson is so fast on ESP8266 (even more so on ESP32) that the second parsing is worth the effort to reduce memory usage.

```
DynamicJsonDocument* getJsonDocFromFile(File f) {
	DeserializationError error;
	boolean final = false;
	size_t sz = ESP.getFreeHeap() - 6*1024; //Free heap space - 6K
	do {
		DynamicJsonDocument *jsonDoc = new DynamicJsonDocument(sz);
		f.seek(0); // make sure file pointer is at beginning of file
		error = deserializeJson(*jsonDoc, f, DeserializationOption::NestingLimit(20));
		if (error) { // Test if parsing fails
			DbgPrintln(F(" - deserializeJson failed: "), error.c_str());
			sz += 256; 
		} else {
			if (final) return jsonDoc; 
			final = true;
			sz = jsonDoc->memoryUsage();
		}
		delete jsonDoc;
	} while ((error == DeserializationError::NoMemory) | final);
	return nullptr;
}

```

Xamtastic, 2019-09-22:
Thanks @ewaldc, I like the work around!

My use case is having a class whose values represent the data in a json settings file, I guess that I could get around the size problem by deserialising JSON with max size similar to how you've done, immediately extracting the values and immediately disposing the DynamicJsonDocument ... and then just recreating it to serialise the values back into json just before saving the file, and then disposing of the object again. The size problem, I suppose, goes away if the objects just exist during serialisation and deserialisation, maybe we've inadvertently created a design pattern problem by wanting the DynamicJsonDocument to persist ... but that problem all goes away if it's existence is just fleeting.

bblanchon, 2019-09-23:
Hi @Xamtastic,

```
error: no match for 'operator[]' (operand types are 'DynamicJsonDocument' and 'std::string')
```

This error means that the support of `std::string` is disabled in `sketch/MyProcessor.cpp`.
You must define [ARDUINOJSON_ENABLE_STD_STRING](https://arduinojson.org/v6/api/config/enable_std_string/) to 1 for every compilation unit.
Right now, it's only on the `.ino` file.

Also, you must preserve `#include <ArduinoJson.h>` in the `.ino` file because that's the way the Arduino IDE detects that you're using the library.

Best Regards,
Benoit

Xamtastic, 2019-09-23:
> You must define [ARDUINOJSON_ENABLE_STD_STRING](https://arduinojson.org/v6/api/config/enable_std_string/) to 1 for every compilation unit.

Thanks @bblanchon, it is surprising how many problems went away with this one, I guess the headline is:

RULE 1: Always put `#define ARDUINOJSON_ENABLE_STD_STRING 1` at the top of every CPP file!

Thanks so much!

closed 2019-09-23 by Xamtastic

ewaldc, 2019-09-24:
It all depends how big the JSON strings are that you have to deal with.  I created an OpenHab 2.x server on ESP which has to deal with JSON strings that require a 48K DynamicJsonDocument.   On the ESP platform there is not enough memory to even serialize it (I use a buffered Print stream writer so I can serialize in 1K chunks).   So I keep the JSON doc in memory.  There is the additional problem that if read your JSON string from SPIFSS, ArduinoJSON will make a copy of the string data in the DynamicJsonDocument.  But the memory of all your object strings is really wasted, since you need to store your values for all your setting in separately allocated memory.  
E.g. if you had initially _"light": "ON"_ than _obj["light"] = "OFF_" will result in _"light": null_   since ArduinoJson has not enough space to store "OFF".   Hence you need to malloc your light state, or if you have fixed states like ON/OFF you can create a pointer to a flash string (but cast it a a non-const!).