bblanchon/ArduinoJson #1095: duplicate key

ilyasamraoui, 2019-09-24:
Hi,
DesrializationError validate json char array that has duplicate key

bblanchon, 2019-09-25:
Hi @ilyasamraoui,

Indeed, the goal of ArduinoJson is to offer a lightweight parser for embedded platforms and not a JSON validator.
I don't think it's worth increasing the size of the code for such an edge case.

Do you have a use case where detecting duplicate keys is crucial?

Best Regards,
Benoit

bblanchon, 2019-09-25:
I just made the test and the code grew by only 26 bytes; so that may be worth it.
But do we really need this?

bblanchon, 2019-09-27:
Looking at [this question on stackoverflow](https://stackoverflow.com/q/21832701/1164966), I realized that duplicate keys are not illegal and are accepted by most implementations.

All major implementations keep the last value:

* Chrome 77: `JSON.parse('{"a":1,"a":2}')` -> `{a: 2}`
* IE 11: `JSON.parse('{"a":1,"a":2}')` -> `{a: 2}`
* Edge 44: `JSON.parse('{"a":1,"a":2}')` -> `{a: 2}`
* Firefox 68: `JSON.parse('{"a":1,"a":2}')` -> `{a: 2}`
* Python 3.6.8: `json.loads('{"a":1,"a":2}')` -> `{'a': 2}`
* Ruby: `JSON.parse('{"a":1,"a":2}')` -> `{"a"=>2}`

So `deserializeJson()` should return `Ok`, but the `JsonDocument` should contain the last value.

ilyasamraoui, 2019-09-27:
ok i see, i used jsonlint.com to validate json object with duplicate key and it didn't validate it

closed 2019-09-30 by bblanchon

bblanchon, 2019-11-01:
Fix published in [ArduinoJson 6.13.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.13.0)

commit a37480ee (2019-09-30): Fixed deserializeJson() when input contains duplicate keys (fixes #1095)