bblanchon/ArduinoJson #1096: JsonObject cannot hold more than few records.

vincentlim5445, 2019-09-25:
Hi, I really need a big help! I’m currently using nodeMCU doing an attendance system project, and using the ArduinoJson to hold all the data in the Firebase. However, if my data is more than few records, it will returns 0 and hence the for loop will not be run. Thank you!

[image]

bblanchon, 2019-09-26:
Hi Vincent,

The [`JsonBuffer`](https://arduinojson.org/v5/api/jsonbuffer/) is running out of memory.
Make sure to destroy it (or at least clear it) between each iteration.

See also:

* [Why shouldn't I use a global JsonBuffer?](https://arduinojson.org/v5/faq/why-shouldnt-i-use-a-global-jsonbuffer/)
* [How to reuse a JsonBuffer?](https://arduinojson.org/v5/faq/how-to-reuse-a-jsonbuffer/)
* [Why parsing fails?](https://arduinojson.org/v5/faq/why-parsing-fails/)

Best Regards,
Benoit

vincentlim5445, 2019-09-26:
Thank you for your reply bblanchon. 

I'm still unable to solve it after changing to the StaticJsonBuffer as follow:
[image]

it show me with the following errors:

`Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

WARNING: library LiquidCrystal_I2C-1.1.2 claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s).
In file included from C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:7:0,

                 from C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:7,

                 from C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\User\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:26,

                 from C:\Users\User\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22,

                 from C:\Users\User\Desktop\Student_CheckIn_Test\Student_CheckIn_Test.ino:6:

C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp: In instantiation of 'struct ArduinoJson::Internals::JsonParserBuilder<ArduinoJson::Internals::StaticJsonBufferBase, const ArduinoJson::JsonVariant, void>':

C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:96:66:   required by substitution of 'template<class TJsonBuffer, class TString> typename ArduinoJson::Internals::JsonParserBuilder<TJsonBuffer, TString>::TParser ArduinoJson::Internals::makeParser(TJsonBuffer*, TString&, uint8_t) [with TJsonBuffer = ArduinoJson::Internals::StaticJsonBufferBase; TString = const ArduinoJson::JsonVariant]'

C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:70:60:   required from 'typename ArduinoJson::Internals::EnableIf<(! ArduinoJson::Internals::IsArray<TString>::value), ArduinoJson::JsonObject&>::type ArduinoJson::Internals::JsonBufferBase<TDerived>::parseObject(const TString&, uint8_t) [with TString = ArduinoJson::JsonVariant; TDerived = ArduinoJson::Internals::StaticJsonBufferBase; typename ArduinoJson::Internals::EnableIf<(! ArduinoJson::Internals::IsArray<TString>::value), ArduinoJson::JsonObject&>::type = ArduinoJson::JsonObject&; uint8_t = unsigned char]'

C:\Users\User\Desktop\Student_CheckIn_Test\Student_CheckIn_Test.ino:223:94:   required from here

C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:73:50: error: no type named 'Reader' in 'struct ArduinoJson::Internals::StringTraits<const ArduinoJson::JsonVariant, void>'

   typedef typename StringTraits<TString>::Reader InputReader;

                                                  ^

C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:74:50: error: no type named 'Reader' in 'struct ArduinoJson::Internals::StringTraits<const ArduinoJson::JsonVariant, void>'

   typedef JsonParser<InputReader, TJsonBuffer &> TParser;

                                                  ^

In file included from C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:7:0,

                 from C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\User\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:26,

                 from C:\Users\User\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22,

                 from C:\Users\User\Desktop\Student_CheckIn_Test\Student_CheckIn_Test.ino:6:

C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp: In instantiation of 'typename ArduinoJson::Internals::EnableIf<(! ArduinoJson::Internals::IsArray<TString>::value), ArduinoJson::JsonObject&>::type ArduinoJson::Internals::JsonBufferBase<TDerived>::parseObject(const TString&, uint8_t) [with TString = ArduinoJson::JsonVariant; TDerived = ArduinoJson::Internals::StaticJsonBufferBase; typename ArduinoJson::Internals::EnableIf<(! ArduinoJson::Internals::IsArray<TString>::value), ArduinoJson::JsonObject&>::type = ArduinoJson::JsonObject&; uint8_t = unsigned char]':

C:\Users\User\Desktop\Student_CheckIn_Test\Student_CheckIn_Test.ino:223:94:   required from here

C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:70:60: error: no matching function for call to 'makeParser(ArduinoJson::Internals::StaticJsonBufferBase*, const ArduinoJson::JsonVariant&, uint8_t&)'

     return Internals::makeParser(that(), json, nestingLimit).parseObject();

                                                            ^

C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:70:60: note: candidate is:

In file included from C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:7:0,

[... 2 lines cut ...]
                 from C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

[... cut ...]
 inline typename JsonParserBuilder<TJsonBuffer, TString>::TParser makeParser(

                                                                  ^

C:\Users\User\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:96:66: note:   substitution of deduced template arguments resulted in errors seen above

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

bblanchon, 2019-09-27:
Hi @vincentlim5445,

You don't have to call `parseObject()` if you already have a `JsonVariant`.

I can see that you are not familiar with ArduinoJson;  please take a moment to read [the documentation on arduinojson.org](https://arduinojson.org/v5/doc/)

Best Regards,
Benoit

closed 2019-10-05 by bblanchon