bblanchon/ArduinoJson #1098: Deserialized data keys not being processed?

Werewoof, 2019-09-29:
Hi there,

I'm quite new to this great little library, so please bear with me. 😓 

I'm trying access nested data in a JSON object. The goal is to make a MIDI program that searches over a set of piano notes and then plays them out to specific channels at specific velocities.
The top-level keys are chord sets: in each of those, there are 4 keys for chords. Within each chord are the notes, whether by formal name or MIDI number (non-HEX).

- I'm running Windows 10 Pro 64bit (if that matters)
- I'm using an Arduino Uno (Chinese clone, but I get same results with a real Arduino Uno)
- I'm compiling with C++11

```
#include <ArduinoJson.h>

const char* chordData = \
  "{"\
  "\"1\":{\"1\":{\"55\":[87,0],\"59\":[56,0],\"62\":[66,0]},\"2\":{\"G3\":[87,0],\"B3\":[56,0],\"D4\":[66,0]},\"3\":{\"62\":[87,0],\"66\":[56,0],\"69\":[66,0]},\"4\":{\"D4\":[87,0],\"F#4\":[56,0],\"A4\":[66,0]}}"\
  ","\
  "\"2\":{\"1\":{\"55\":[87,0],\"59\":[56,0],\"62\":[66,0]},\"2\":{\"G3\":[87,0],\"B3\":[56,0],\"D4\":[66,0]},\"3\":{\"62\":[87,0],\"66\":[56,0],\"69\":[66,0]},\"4\":{\"D4\":[87,0],\"F#4\":[56,0],\"A4\":[66,0]}}"\
  "}";

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  DynamicJsonDocument doc(1024);
  deserializeJson(doc,chordData);
  
  // JsonObject root = doc[String(1)][String(2)].as<JsonObject>(); <--- DOESN'T WORK
  JsonObject root = doc[String(2)][String(2)].as<JsonObject>(); // WORKS FINE
  for (JsonPair kv : root ) {
      Serial.println(kv.key().c_str());
      
      JsonArray array = root[kv.key().c_str()].as<JsonArray>();
      for(JsonVariant v : array) {
          Serial.println(v.as<int>());
      }
  }
  delay(2000);
}
```
**Output (doc[String(1)][String(2)])**
>  *nothing*

**Output (doc[String(2)][String(2)])**
> G3
87
0
B3
56
0
D4
66
0
G3
87
0
B3
56
0
D4
66
0

I'm guessing the issue _could_ be due to insufficient memory, but it's possible I'm doing something wrong, or un-ideally.

Any help would be greatly appreciated!

bblanchon, 2019-09-30:
Hi @Werewoof,

This looks like a memory problem. You can confirm by printing the `String`, or checking [`DynamicJsonDocument::capacity()`](https://arduinojson.org/v6/api/jsondocument/capacity/).

Anyway, my rule is:

> Arduino UNO == no heap memory

Start by replacing the [`DynamicJsonDocument`](https://arduinojson.org/v6/api/dynamicjsondocument/) with a [`StaticJsonDocument`](https://arduinojson.org/v6/api/staticjsondocument/).
Then get rid of all the `String`s and use `char[]` instead.

Remember that the UNO only has 2KB of RAM, so a `JsonDocument` of 1024 can work but it rather large. Use the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) to compute the right capacity.

Finally, you can replace:

```c++
JsonArray array = root[kv.key().c_str()].as<JsonArray>();
```

with:

```c++
JsonArray array = kv.value();
```

This won't reduce memory usage by will safe a useless lookup.

Best Regards,
Benoit

Werewoof, 2019-09-30:
Hi again,

Thank you so much for your help! All of the changes are working smoothly. I can see now, this was definitely a memory error: I'm going to make some changes to make my code more efficient. 

Thanks again :)

closed 2019-09-30 by Werewoof

bblanchon, 2019-10-01:
You're welcome @Werewoof.
Good luck with your project.
Don't forget to star the repository to support the project ;-)