bblanchon/ArduinoJson #1099: Question Use deserializeJson with Wire

quantumlicht, 2019-10-02:
Hello, first of all, thanks for the amazing lib!

I am trying to have 2 chips communicate with the Wire protocol (NodeMCU master and Arduino Uno as slave)

It works fine when I send simple string messages between the two. However, when I throws arduinoJson in the mix, I am running into issues. from the docs, I can read that Wire is supposed to be supported for `deserializeJson`, However I can't seem to make it work. It appears that the serialization on the slave is working just fine:

I am using version **6.12.0**. I have not tried other versions yet, but I definitely can if you think there were recent changes that may have affected this.

# Master (NodeMCU)
```c
const size_t capacity = JSON_OBJECT_SIZE(3) + 30;
void setup(){
  Wire.begin(D1, D2); /* join i2c bus with SDA=D1 and SCL=D2 of NodeMCU */
  Serial.begin(9600); /* begin serial for debug */
  Serial.println("setup::done");
}

void loop() {
  delay(50);
  Wire.beginTransmission(8); 
  Wire.write("Hello Arduino");
  Wire.endTransmission(); 

  Wire.requestFrom(8, capacity);
  DynamicJsonDocument doc(capacity);

  DeserializationError err = deserializeJson(doc, Wire);
  switch (err.code()) {
    case DeserializationError::Ok:
        Serial.print(F("Deserialization succeeded"));\
        serializeJson(doc, Serial);
        break;
    case DeserializationError::InvalidInput:
        Serial.print(F("Invalid input!"));
        break;
    case DeserializationError::NoMemory:
        Serial.print(F("Not enough memory"));
        break;
    default:
        Serial.print(F("Deserialization failed"));
        break;
  }
}
```
Which goes into the `DeserializationError::InvalidInput` case.

# Slave (Arduino Uno)
```c
const size_t capacity = JSON_OBJECT_SIZE(3);
void setup(){
   Wire.begin(8);
  Wire.onReceive(receiveEvent); /* register receive event */
  Wire.onRequest(requestEvent); /* register request event */
  Serial.begin(9600);
}
void dumpMetrics(){  
  DynamicJsonDocument root(capacity);

  root["ts_v"]= "10";
  root["ts_t"] = "gauge";
  root["ts_c"] = "ul";

  
  serializeJson(root, Wire);
  serializeJson(root, Serial);
  Serial.println();

void requestEvent() {  
  dumpMetrics();
 }
```
Which prints `{"ts_v":"10","ts_t":"gauge","ts_c":"ul"}` to Serial as you would expect.

I am doing something wrong? I am kind of a noob in embedded, so I might be missing something important.

Let me know if you need more info.

bblanchon, 2019-10-02:
Hi @quantumlicht,

The Wire library implements the `Stream` interface, so it should be compatible with ArduinoJson. I never tried but I can probably find two Arduinos if you're really stuck.

Meanwhile, you should check what `deserializeJson()` receives by using the `LoggingStream` decorator from [StreamUtils](https://github.com/bblanchon/ArduinoStreamUtils):

```c++
LoggingStream loggingWire(Wire, Serial);
DeserializationError err = deserializeJson(doc, loggingWire);
```

`LoggingStream` will print every byte to the serial port, so you'll be able to see what `deserializeJson()` receives.

Best regards,
Benoit

quantumlicht, 2019-10-02:
Merci Benoît!

I will definitely checkout the StreamUtils and keep you posted on how this goes. 
I only see one reference to StreamUtils in the docs, it would be nice to make it more prominent in the documentation :)

bblanchon, 2019-10-16:
Hi @quantumlicht,

Did you make some progress?

quantumlicht, 2019-10-16:
Unfortunately I decided to not use Json and go with a much simpler message passing protocol.
I had a really hard time working with the Wire library. I am not sure streams work well. I think it has to do with the size of the receiving buffer. Thanks for following up. No action required on your part (e.g. You can close this ticket), but I wonder if anyone has made it work and if so, how.

closed 2019-11-21 by bblanchon