bblanchon/ArduinoJson #1103: IncompleteInput when deserializing a string of complex nested JSON object

openedhardware, 2019-10-07:
Device: **Particle Photon**
Platform: **VS Code with Particle Workbench**

I am trying to parse this JSON object - https://jsonblob.com/0e38cfb3-b35f-11e9-8c40-8f8bdf2615fc

Here is my code:
```
  String str_body = ".........";  // Long string from https://jsonblob.com/0e38cfb3-b35f-11e9-8c40-8f8bdf2615fc
  Serial.print("\tLength: "); Serial.println(str_body.length());
  Serial.println(str_body);
  char response_char[4096];
  str_body.toCharArray(response_char, str_body.length());
  DeserializationError error = deserializeJson(doc, response_char);
  if (error) {
    Serial.println(F(">>>>> Failed to deserialize the response from the server."));
    Serial.println(error.c_str());
    return;
  }
  config_obj = doc.as<JsonObject>();
```

But I am getting this error:
```
### Response status: 200	Length: 878
{"rules":[{"input":{"type":"digital","condition":{"parameter":"D1","condition":"equal","value":"ON"}},"commands":[{"type":"sms","parameter":"250-7070","value":"Silo Interlock activated"},{"type":"relay","parameter":"1","value":"ON","delay":0},{"type":"email","parameter":"a@a.com","value":{"subject":"Bolier Warnning!","text":"Boiler 23 needs attention"}}]},{"input":{"type":"analog","condition":{"parameter":"A1","condition":"greater than","value":"0"}},"commands":[{"type":"sms","parameter":"250-7070","value":"Silo Interlock activated"},{"type":"relay","parameter":"2","value":"OFF","delay":3},{"type":"email","parameter":"a@a.com","value":{"subject":"Bolier Warnning!","text":"Boiler 23 needs attention"}}]},{"input":{"type":"sms","condition":{"condition":"equal","value":"12345=789"}},"commands":[{"type":"sms","parameter":"250-7070","value":"Silo Interlock activated"}]}]}
>>>>> Failed to deserialize the response from the server.
IncompleteInput
```

Any idea?

closed 2019-10-07 by openedhardware

bblanchon, 2019-10-08:
Hi @openedhardware,

I'm assuming you fixed your problem.
What was the solution?

Best Regards,
Benoit

openedhardware, 2019-10-08:
```
// str_body.toCharArray(response_char, str_body.length());
str_body.toCharArray(response_char, str_body.length() + 1);
```