bblanchon/ArduinoJson #1104: Can't create an int64_t field in StaticJsonDocument

pabloVecchio, 2019-10-07:
Hi, i'm trying to implement your library (v6), but i can't figure out how to create a int64_t field. In this function:

bool Class::Function ( int64_t value, String text )  {
	StaticJsonDocument<500> doc;
	doc["value"] = value;
	doc["text"] = text;

	String msg;
	serializeJson(doc, msg);
	return workWith(msg);
}

doc["value"] is always cast to 32 bits, even defining: 
#define ARDUINOJSON_USE_LONG_LONG 1
#include "ArduinoJson.h"     // there is no other include for ArduinoJson.h

There is a way to create a json with a int64_t field ?

Thanks!

bblanchon, 2019-10-08:
Hi @pabloVecchio,

Which platform are you targetting?

Best Regards,
Benoit

pabloVecchio, 2019-10-08:
Sorry, i didn't say that, it's a ESP8266. I can read the int_64 value doing:
int64_t var = doc["key"]; // ie: "key": 68719476735 //(0xFFFFFFFFF)
But not write

bblanchon, 2019-10-09:
The library works fine on PC, as you can see on this demo:
https://wandbox.org/permlink/ksXpcoDQL0n3BXeo

So there might be something wrong with ESP8266.
Let me try to reproduce the issue...

bblanchon, 2019-10-10:
Hi @pabloVecchio,

I just tried on an ESP8266, and it works like a charm when [ARDUINOJSON_USE_LONG_LONG](https://arduinojson.org/v6/api/config/use_long_long/) is set to 1.

I guess that you have several compilation units (several `.cpp`/`.ino`), but only defined ARDUINOJSON_USE_LONG_LONG for one of them. You must use the same value in each compilation unit; otherwise, your program will embed two variants of the library: one with `long long` and one without.

Best Regards,
Benoit

pabloVecchio, 2019-10-12:
You're right!! I forgot that i have a ' #include "ArduinoJson.h" ' in another file, added ARDUINOJSON_USE_LONG_LONG there too and is working!
Thanks a lot for your time!!

bblanchon, 2019-10-12:
You're welcome @pabloVecchio.
Good luck with your project.
And don't forget to [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show your support 😉

closed 2019-10-12 by bblanchon