bblanchon/ArduinoJson #1105: cannot include base64 image string to json

ligantx, 2019-10-08:
i use esp32-cam and i want to include an image in base64 encoding to a json i create..
if i copy a base64 long image string inside code it works fine:
```
Serial.println(base64image); //shows the current base64 image String correctly
JsonObject requests_0_image["content"];
requests_0_image["content"] = "/9j/4AA....long base64 String..."
serializeJson(doc, myJsonString);
Serial.println(myJsonString); //Json is fine
```

if i try:
```
Serial.println(base64image); //shows the current base64 image String correctly
JsonObject requests_0_image["content"];
requests_0_image["content"] = base64image;
serializeJson(doc, myJsonString);
Serial.println(myJsonString); //Json is like {"requests":[{"image":{"content":null},.....
```
is also tried:
`requests_0_image["content"] = serialized(base64image);
`
but its also doesnt work..
important: if i use a small String as base64image the code works fine..
so, i guess it has to do with String length (i read that " the content of the String will be duplicated in the JsonDocument" so i guess it is causing the problem)

bblanchon, 2019-10-09:
Hi @ligantx,

Indeed ArduinoJson duplicates `char*` and `String`s in the `JsonDocument`.
To avoid this duplication, you must pass a `const char*`.

In your case, you can do:

```c++
requests_0_image["content"] = base64image.c_str();
```

`String::c_str()` returns a `const char*`  pointing to the internal buffer of the `String`.
This solution works as long as you keep the `String` in memory, and you don't modify it.

Best Regards,
Benoit

ligantx, 2019-10-09:
i'll try that and i'll post the results..
until now i found a workaround that works (after many many hours trying to figure this out) and maybe help others:
make a function that will create and return the whole json except the "content" value.
Then i substring the returned json (using indexOf(content) + offset) concatenate the create base64image String and then concatenate the rest of the json (the substring from the previous indexOf till the end).
So, i finally get a big Json that includes the image, without duplicated anything..

closed 2019-11-01 by bblanchon