bblanchon/ArduinoJson #1107: Create a key with value null results in wrong output

m-andersen, 2019-10-09:
I needed to serialize a JSON document and I'm doing all the usual stuff which works.
I then needed a key where the value is set to null. I checked the documentation how to do that and one way was
 doc["me"].add((char*)0);

which results in this output:
"me": [null]

This is not what I expected. I expected
"me": null

After more searching around I found this is also valid
 doc["me"] = (char*)0;

which results in the wanted output:
"me": null

Please evaluate the documentation or code on how to add a null-value as the obj.add() method gives wrong output

m-andersen, 2019-10-13:
> Hi @m-andersen <https://github.com/m-andersen>,
>
> It's part of the JsonDocument magic.
> Unlike JsonVariant, JsonDocument changes its type to match to the way you use it:
>
> use operator[] and it becomes an object;
> use add() and it becomes an array.
> For example, you wrote doc["me"], so the root of the document becomes an object.
> Then, you wrote doc["me"].add(nullptr), so the object member "me" becomes an array.
>
> This feature is documented in the section Using a JsonDocument <https://arduinojson.org/v6/api/jsondocument/> but it may not be easy to find.
> Do you have any suggestion? Where did you look?
>
> Best Regards,
> Benoit

Hi

In the book chapter 4.3.8 it is mentioned to use .add to add a value of null, but this results in wrong output as mentioned. This chapter should be updated with this example instead

doc["me"] = (char*)0;

to get
"me": null

br
Martin

bblanchon, 2019-10-14:
Hi Martin,

Section 4.3 covers **arrays**.

The example is:

```c++
arr.add((char*)0);
```

In this example, `arr` is a `JsonArray`, not a `JsonObject`.

But, you're right, if it were a `JsonObject`, we would have to write:

```c++
doc["key"] = (char*)0;
```

Best Regards,
Benoit

m-andersen, 2019-10-14:
Ok, it was not clear arr was an array as the text describes this as a JSON document. I suggest to update it to be more clear how to add it for a document and how to add it for an array (existing example).

Martin

bblanchon, 2019-10-15:
Alright, thanks for the suggestion Martin.

closed 2019-10-15 by bblanchon