bblanchon/ArduinoJson #1108: Non-null String is serialising as null

tonilopezmr, 2019-10-10:
I'm trying to serialize a JSON object with a simple `String`. If I print it using `Serial` it's just fine, but when I `serializeJson` the result is `null`.

- I have tried with `.c_str()` and then I get weird ASCII simbols.
- I have tried to hardcode a simple string `"boardName"` and its working, it seems to be something related to that `String` but Serial can handle and `ArduinoJson` can't.

```c++
String getBoardName() {
    char buffer[20];
    snprintf_P(buffer, sizeof(buffer), PSTR("%s-%06X"), "BOARD", ESP.getChipId());
    return String(buffer);
}

DynamicJsonDocument getStatus() {
    const size_t capacity = JSON_OBJECT_SIZE(2);
    DynamicJsonDocument root(capacity);

    root["relay"] = getTransistorStatus();
    String boardName = getBoardCode();
    Serial.println("Boardname: "+ boardName);
    root["boardName"] = boardName;
    
    serializeJson(root, Serial);
    return root;
}
```

Output:

```
BOARD-23423
{"boardName": null}
```

Some idea? Thanks!

tonilopezmr, 2019-10-10:
wow, thanks! I didn't see that FAQ question. Let me check and I'll close this PR. 🎉

ThisIsAlsoMyPassword, 2019-10-10:
Hi,
I would like to expand on that question because my problem is very similar but not really solved by the FAQ, I think.
I have an issue with changing a field within a `DynamicJsonDocument` in place using the assignment operator like so:
```c++
const size_t capacity = JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(4) + 46;
DynamicJsonDocument doc(capacity);
doc["time"]="111111111111"
char buf[15]
loop(){
buf = "222222222222";
doc["time"] = buf;}
serializeJson(doc, Serial);
```
After several changes, the field gets serialized as "null" like with the original example. If the document capacity is increased it just yields "null" later.
I'm not sure if this is an issue or a misunderstandig on my part, that's why I posted it first [here](https://forum.arduino.cc/index.php?topic=639500.msg4328291#msg4328291) with the complete test code and output.
Cheers

tonilopezmr, 2019-10-10:
You are right @bblanchon, I used the assistant but don't understand why then should I use more capacity than the `JSON_OBJECT_SIZE` give me if I use only two small objects (not heavy one).

@ThisIsAlsoMyPassword It seems you are doing a loop in the middle, then you are increasing the capacity.

My recommendation then always uses always some MAX number like in this case `150` or something like that to not have this kind of surprises.

Thanks!  @bblanchon that fixed my problem, I close the issue 👍

closed 2019-10-10 by tonilopezmr

bblanchon, 2019-10-10:
@ThisIsAlsoMyPassword, your program makes a new copy of the string at each iteration.
Because you don't reset the `JsonDocument` between each iteration, it fills up until it's full.
To solve this problem, you just need to move the declaration of `doc` inside the loop.

See also:

* [Why shouldn't I use a global JsonBuffer?](https://arduinojson.org/v5/faq/why-shouldnt-i-use-a-global-jsonbuffer/) (it's for ArduinoJson 5, but most of it still applies).