bblanchon/ArduinoJson #1109: missing data when parsing

red-kooga, 2019-10-11:
Hi, is there a different way to parse json when the format is:

mosquitto_pub -t esp/test1 -m '{"Time":"2019-10-11T14:11:02","SR04":{"Distance":26.895}}'

compared to:

mosquitto_pub -t esp/test1 -m '{"Time":"2019-10-11T14:11:02","Distance":26.895}'

With "SR04" not working

14:32:18.982 -> 2019-10-11T14:11:02
14:32:18.982 -> error :Ok
14:32:18.982 -> sensor : 0.00 .

When i remove "SR04" works

14:32:48.966 -> sensor : 26.89 .
14:33:08.972 -> 2019-10-11T14:11:02
14:33:08.972 -> error :Ok

I can also parse both correctly in node-red.

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

const char* ssid = "ssid";
const char* password =  "passwd";
const char* mqttServer = "192.168.1.5";
const int mqttPort = 1883;
const char* mqttUser = "usr";
const char* mqttPassword = "passwd";

WiFiClient espClient1;
PubSubClient client(espClient1);

void setup() {

  Serial.begin(115200);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");

  client.setServer(mqttServer, mqttPort);
  client.setCallback(callback);

  while (!client.connected()) {
    Serial.println("Connecting to MQTT...");

    if (client.connect("espClient1", mqttUser, mqttPassword )) {
      Serial.println("connected");
    }
    else {
      Serial.print("failed with state ");
      Serial.print(client.state());
      delay(2000);
    }
  }
  client.subscribe("esp/test1");
}
void callback(char* topic, byte* payload, unsigned int length) {

  DynamicJsonDocument doc(256);
  DeserializationError error = deserializeJson(doc, payload, length);
 
  Serial.println(doc["Time"].as<char*>());
  Serial.print("error :");
  Serial.println(error.c_str());

  float sensor = doc["Distance"];
  Serial.print("sensor : ");
  Serial.print(sensor);
  Serial.println(" .");
}

void loop()

{
  delay(5000);
  client.loop();
}

Thanks

mikee47, 2019-10-12:
Try
```
float sensor = doc["SR04"]["Distance"];
```
or
```
JsonObject sr04 = doc["SR04"];
float sensor = sr04["Distance"];
```

bblanchon, 2019-10-13:
Hi @red-kooga, 

I'd say that your `JsonDocument` is too small.

However, I don't understand why the [DeserializationError](https://arduinojson.org/v6/api/misc/deserializationerror/) is still `Ok`; it should be `NoMemory`.
It might be a bug, can you double-check?

If you confirm that [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) still returns `Ok` instead if `NoMemory`, please create an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) that I can run offline.

Best Regards,
Benoit

red-kooga, 2019-10-13:
I'll test it Tuesday when I get home

Thanks

red-kooga, 2019-10-17:
Hi mikee47, both your suggestions work
Thanks

red-kooga, 2019-10-17:
Hi bblanchon, DeserializationError is still ok, occasionally get deserializeJson() failed: InvalidInput when first connect to mqqt server, but rock solid after that
Thanks

bblanchon, 2019-10-17:
If @mikee47 suggestions worked, it means that I make the wrong diagnostic.
There is probably nothing wrong with the library.
Thanks for the update @red-kooga.

closed 2019-10-17 by bblanchon