bblanchon/ArduinoJson #1110: Feature request: XML serialisation

mikee47, 2019-10-12:
Are there any plans to incorporate this into the library? A particular application I have in mind is generating UPNP description documents, which are in XML format, using ArduinoJson as the data source.

It would probably be quite easy to add this as an alternative serialiser/deserialiser (http://convertjson.com/json-to-xml.htm) but wondering if it had been considered as an extension?

P.S. Very impressive library, very useful indeed. Thank you!

bblanchon, 2019-10-12:
Hi @mikee47,

I think that's possible to write a `serializeXml()`, but writing a `deserializeXml()` would be a nightmare (except, maybe, if we limit ourselves to a subset of XML, like Plist).

But the real question is: how many users would be interested?
My guess is: very few, and probably just one.
As a comparison, I added support for MessagePack a year ago, but it represents only **0.30%** of the traffic on arduinojson.org.

Best Regards,
Benoit

mikee47, 2019-10-12:
Hi Benoit, thanks for responding so quickly! I agree with your comments, especially about deserialisation although things like https://github.com/leethomason/tinyxml2 would probably make the job easier.

As to whether any users would be interested, that's probably impossible to tell. Take JSON as an example; it's very easy to hack together code to create valid JSON, not so much more difficult to have a go at parsing it. And many users probably thought that was fine until they discovered ArduinoJson!

I'll probably have a stab at the serializer anyway, but if you're not really interested I won't bother with a PR here.
Regards,
Mike.

mikee47, 2019-10-24:
I've found a better solution using RapidXML, so I'll close this issue. Thanks.

closed 2019-10-24 by mikee47