bblanchon/ArduinoJson #1112: Cannot build a project with #include "ArduinoJson.h" in Atmel Studio 7

mgcavallo, 2019-10-17:
Hi,

I am not able to build a project in Atmel Studio 7 which include arduinoJson library. When I buid it, I get errors in relation with the #include directives that are inside the own library. Whta's more, I created a new project in which I only included the arduinoJson library and I left the main function empty and I got the same errors. I tried to add the library in the different ways, but the result is same.

I mean, the compiler does not find the files. The error descriptions say "xxxxxxx.h:No such file or directory." 

In some case, it seems like the solution in add .h extension into the name of the file, but it is not the solution in other cases.

Have anybody got an idea? I attach some pictures for you.

Thanks in advance.

[image]
[image]
[image]
[image]
[image]

bblanchon, 2019-10-17:
Hi @mgcavallo,

ArduinoJson looks for `<string>` and `<ostream>` because it incorrectly guessed that the target is a computer (as opposed to a microcontroller).

Try to set `ARDUINOJSON_EMBEDDED_MODE` to 1:

```c++
#define ARDUINOJSON_EMBEDDED_MODE 1
#include "ArduinoJson-v6.12.0.h"
```

This will disable support for `std::string` and `std::ostream`, which should fix your problem.

There is probably a way to correctly detect the target type, I'll have a look ASAP.

Best Regards,
Benoit

mgcavallo, 2019-10-17:
Hi @bblanchon ,

Thank you for your prompt response.
That's right! The #define directive setting ARDUINOJSON_EMBEDDED_MODE to 1, resolve my issue.

However, I agree with you that there would be a way to detect the target type.

Best Regards,
Martin

mgcavallo, 2019-10-18:
Hi, 

I am having a new issue. I think it is a silly issue, but I do not unerstand.
I am migrating from ArduinoJson5 to ArduinoJson6 and I have the problem with serialization. The error said that no match function for the arguments. I want to serialize a StaticJsonDocument into a char array. The documentation shows that it is possible.

Attach pictures.

[image]
[image]

mgcavallo, 2019-10-18:
Hi,

I could solve the problem adding the lenght as the third parameter.

Regards,

Martin

closed 2019-10-19 by bblanchon

bblanchon, 2019-11-01:
Fix published in [ArduinoJson 6.13.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.13.0)

mgcavallo, 2019-11-05:
Thank you Benoit!!! I will try verison 6.13.0 ASAP.

commit d4f819f1 (2019-10-19): Added detection of Atmel AVR8/GNU C Compiler (fixes #1112)