bblanchon/ArduinoJson #1114: ArduinoJson assistant only part of the  input code apears in the "Parsing Program" section

aardkop, 2019-10-21:
I have a verry large Json output from Domoticz but in the "parsing program" section there is only a part of the code. in "the serialising" section it is complete. 
Is there a restricition on the size?

Regads Adriaan

bblanchon, 2019-10-22:
Hi Adriaan,

Yes, the ArduinoJson Assistant only shows the four first elements of each array.
I thought it was enough to show how the library works; I didn't realize it would scare some users ;-)
Do you think I should remove the restriction?

Best Regards,
Benoit

aardkop, 2019-10-22:
Dear Benoit,

Im building a e-ink screen for my domoticz, and the json output is quite big. im not a big coder and use scripts from all over the web, your tool works perfect for me but the restriction is holding me back. for me personally it would be verry usefull if i can use it with a large file 😁.

Kind regards,

Adriaan.

p.s. thank you for the great work, it helps a noob coder like me a lot!

bblanchon, 2019-10-22:
I removed the limitation to four elements; however, you should really consider writing a loop.
See [JsonArray::begin() / JsonArray::end()](http://127.0.0.1:4000/v6/api/jsonarray/begin_end/)

aardkop, 2019-10-22:
Thank you Benoit, i will take a look at your pointer.

closed 2019-10-22 by aardkop