bblanchon/ArduinoJson #1115: Strings memory leak

Avenitos, 2019-10-21:
Hello! I just can’t understand why memory leaks in JsonDocument?

`#include <Arduino.h>
#include <ArduinoJson.h>

DynamicJsonDocument doc(256);

void setup() {
  Serial.begin(9600);
}

void SetValue(String Value) { 
  doc["name"].set(Value);
}

void loop() {
  SetValue("value");

  Serial.println(String(doc.memoryUsage()) + " " + String(doc.capacity()) + " " + String(doc.size()));  

  String s;
  serializeJson(doc, s);
  Serial.println(s);

  delay(1000);
}`

Serial output:
`250 256 1
{"name":"value"}
256 256 1
{"name":"value"}
256 256 1
{"name":null}
256 256 1
{"name":null}
256 256 1
{"name":null}
256 256 1
{"name":null}`

bblanchon, 2019-10-22:
Hi @Avenitos.

I answered your question in the FAQ: [I found a memory leak in the library!](https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/)

Best Regards,
Benoit

Avenitos, 2019-10-22:
Thank you for your answer!

I tried out these ways yesterday... But I don 't see the possibility of applying them in my project, so I asked a question. My project uses a global JsonDocument to store data, loads it from SPIFFS when it starts up, and saves it periodically.

Will you have to abandon ArduinoJson and do your dynamic structure?

Maybe there 's another way out?

bblanchon, 2019-10-23:
Hi @Avenitos,

ArduinoJson is a serialization library.
Are you using ArduinoJson for serialization?
No, you're using it as a generic container for the state of your application, something ArduinoJson was not designed for.

Instead, you must create custom data structures, as shown in the  [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/) example.

See also:

* [Why must I create a separate config object? Why can't I use JsonDocument directly?](https://arduinojson.org/v6/faq/why-must-i-create-a-separate-config-object/)

Best Regards,
Benoit

ewaldc, 2019-10-25:
@Avenitos,
My application is doing something similar as yours.  It was running on (a slightly modified) ArduinoJson 5.x happily and was also using the ArduinoJson (dynamic) buffer somewhat as a context structure for application state: an representation of a hierarchy of IOT devices (sensors, switches, sliders etc.) containing fixed properties like name, type etc. but also their state, all of which is initially read from SPIFFS. Once every week the objects are written to flash (SPIFSS) and the ESP reset.
Back then I was not aware (maybe I missed it...) than such usage **might** not really be supported.

I was aware though of the fact that ArduinoJson would duplicate a string or a constant (e.g. in flash) and thus at each value change (e.g. switch moving from ON to OFF continuously) would "leak" memory.
So I kept the actual state data as (class) structures (as recommend in the V6.x FAQ) and having the initial JSON values of the state objects as null ("state": "") so that ArduinoJson would not allocate initial space for it.  It is using a _setState_ function that initializes and then manages all memory as the state changes (in reality these are implemented as class setter/getter for each object type e.g. a "switch").  For constants, like switch "ON"/"OFF", they can't be in flash given the alignment issues (actually my modified ArduinoJSON 5.x implemented zero-copy flashstrings) but if stored globally or in a class private storage as a _static char*_ they anyhow would, in the case of a switch, only consume 7 bytes of RAM for all switches combined. 
```
void setState(Item *item, char *state) {
	uint_fast16_t len = strlen(state) + 1;
	item->state = (char *)realloc(item->state, len);
	memcpy(item->state, state, len);
	item->obj[F("state")] = (char*)item->state;
}
```
When I ported to ArduinoJson 6 however, I discovered the new FAQ's/guidelines regarding this type of usage.  
As of now (and not willing to redesign the whole code) , I am also still struggling with some memory leaks that I have not yet found.  My code is compliant though with the JsonConfigFile.ino example (managed state data outside of _JsonDocument_), but I'm not totally sure if I have not missed any other recommendation.
It's possible these leaks may not be caused by ArduinoJSON as I already found a few memory leaks in ESP8266 libraries (most fixed now in 2.6.0-DEV).  Fortunately, all is still running fine on V5.11 so I am not pressured by time...  Will leverage your little test code to see if it leaks on the side of AJ 6.12.

closed 2019-11-21 by bblanchon