bblanchon/ArduinoJson #1116: How can I make this dynamic

gabi68, 2019-10-26:
I am quite new on this. I found arduinojson5 assistant  and that give a memory pool size and some code.
That code is char based only (almost hardcoded), but I need that to be dynamic. What I mean by that is to have the ability to input an address and have that specific input which can be used later to print different values.

How can I do that?

Thank you
Gabi
[code]
/*************************************(openweathermap.org)**************************************/

char servername1[]="eu1.locationiq.com:443"; 
void getlocation()                                //client function to send/receive GET request data.
   {
   if (client.connect(servername1, 80))   
          {                                         //starts client connection, checks for connection
          //client.stop();
          client.println("GET /eu1.locationiq.com/v1/search.php?key=b93717f9237d52&q=" + addresscode +  " &format=json"  ); 
          //"Host: eu1.locationiq.com:443"
          //GET //eu1.locationiq.com/v1/search.php?key=b93717f9237d52&q=29 mason road, crawley, rh10 6dn&format=json
          client.println("Host: eu1.locationiq.com: 443");
          client.println("User-Agent: ArduinoWiFi/1.1");
          client.println("Connection: close");
          client.println();
          } 
  else {
         Serial.println("connection failed");        //error message if no client connect
          Serial.println();      
       }  
  while(client.connected() && !client.available()) 
  delay(1);                                          //waits for data
 
 const size_t capacity = JSON_ARRAY_SIZE(1) + JSON_ARRAY_SIZE(4) + JSON_OBJECT_SIZE(11) + 370;
DynamicJsonBuffer jsonBuffer(capacity);

//const char* json = "[{\"place_id\":\"103835827\",\"licence\":\"https://locationiq.com/attribution\",\"osm_type\":\"way\",\"osm_id\":\"95927357\",\"boundingbox\":[\"51.1323134\",\"51.1327891\",\"-0.0226613\",\"-0.0217598\"],\"lat\":\"51.1323134\",\"lon\":\"-0.0222589\",\"display_name\":\"Engalee, East Grinstead, Mid Sussex, West Sussex, South East, England, RH19 1FE, United Kingdom\",\"class\":\"highway\",\"type\":\"residential\",\"importance\":0.6}]";

JsonArray& root = jsonBuffer.parseArray(json);

JsonObject& root_0 = root[0];
const char* root_0_place_id = root_0["place_id"]; // "103835827"
const char* root_0_licence = root_0["licence"]; // "https://locationiq.com/attribution"
const char* root_0_osm_type = root_0["osm_type"]; // "way"
const char* root_0_osm_id = root_0["osm_id"]; // "95927357"

JsonArray& root_0_boundingbox = root_0["boundingbox"];
const char* root_0_boundingbox_0 = root_0_boundingbox[0]; // "51.1323134"
const char* root_0_boundingbox_1 = root_0_boundingbox[1]; // "51.1327891"
const char* root_0_boundingbox_2 = root_0_boundingbox[2]; // "-0.0226613"
const char* root_0_boundingbox_3 = root_0_boundingbox[3]; // "-0.0217598"

const char* root_0_lat = root_0["lat"]; // "51.1323134"
const char* root_0_lon = root_0["lon"]; // "-0.0222589"
const char* root_0_display_name = root_0["display_name"]; // "Engalee, East Grinstead, Mid Sussex, West Sussex, South East, England, RH19 1FE, United Kingdom"
const char* root_0_class = root_0["class"]; // "highway"
const char* root_0_type = root_0["type"]; // "residential"
float root_0_importance = root_0["importance"]; // 0.6
root_0["lat"].printTo(Serial);

}
// =======================================================================
[/code]

bblanchon, 2019-10-27:
Hi @gabi68,

I'm not sure what your question is...
Are you asking how to make an HTTP request?
If so, look at the example [JsonHttpClient.ino](https://arduinojson.org/v6/example/http-client/)

Best Regards,
Benoit

gabi68, 2019-10-28:
Hi Benoit,

Can you a have a look on this and tell me why this failed to parse?

const size_t capacity = JSON_ARRAY_SIZE(1) + JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(2) + 7 * JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(7) + JSON_OBJECT_SIZE(15) + 900;
  DynamicJsonBuffer jsonBuffer(capacity);
  JsonObject& root = jsonBuffer.parseObject(client);
  if (!root.success()) {
    Serial.println("Parsing failed!");
    return;
  }
  lat = root["GeoPosition"]["Latitude"];
  lon = root["GeoPosition"]["Longitude"];

  Serial.println("Response:");
  Serial.println(lat, 3); //Leh
  Serial.println(lon, 3);
  client.stop();
The const behind 
(const char* json = [{\"Version\":1,\"Key\":\"715290\",\"Type\":\"City\",\"Rank\":85,\"LocalizedName\":\"Crawley\",\"EnglishName\":\"Crawley\",\"PrimaryPostalCode\":\"RH10 1\",\"Region\":{\"ID\":\"EUR\",\"LocalizedName\":\"Europe\",\"EnglishName\":\"Europe\"},\"Country\":{\"ID\":\"GB\",\"LocalizedName\":\"United Kingdom\",\"EnglishName\":\"United Kingdom\"},\"AdministrativeArea\":{\"ID\":\"WSX\",\"LocalizedName\":\"West Sussex\",\"EnglishName\":\"West Sussex\",\"Level\":1,\"LocalizedType\":\"Non-Metropolitan County\",\"EnglishType\":\"Non-Metropolitan County\",\"CountryID\":\"GB\"},\"TimeZone\":{\"Code\":\"GMT\",\"Name\":\"Europe/London\",\"GmtOffset\":0,\"IsDaylightSaving\":false,\"NextOffsetChange\":\"2020-03-29T01:00:00Z\"},\"GeoPosition\":{\"Latitude\":51.113,\"Longitude\":-0.186,\"Elevation\":{\"Metric\":{\"Value\":76,\"Unit\":\"m\",\"UnitType\":5},\"Imperial\":{\"Value\":249,\"Unit\":\"ft\",\"UnitType\":0}}},\"IsAlias\":false,\"SupplementalAdminAreas\":[{\"Level\":0,\"LocalizedName\":\"England\",\"EnglishName\":\"England\"},{\"Level\":2,\"LocalizedName\":\"Crawley\",\"EnglishName\":\"Crawley\"}],\"DataSets\":[\"Alerts\",\"MinuteCast\",\"Radar\"]}]";)

The board is a Wemos D mini
Many thanks in advance
Gabi

ewaldc, 2019-10-28:
@gabi68,
this is a Json Array IMHO. 
Hence, parse with
`JsonArray& root = jsonBuffer.parseArray(client);`

Ewald

gabi68, 2019-10-28:
@ewaldc 
Still "Parsing failed!" 

New code:

const size_t capacity = JSON_ARRAY_SIZE(1) + JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(2) + 7 * JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(7) + JSON_OBJECT_SIZE(15) + 900;
  DynamicJsonBuffer jsonBuffer(capacity);
  //JsonObject& root = jsonBuffer.parseObject(client);
  JsonArray& root = jsonBuffer.parseArray(client);
  if (!root.success()) {
    Serial.println("Parsing failed!");
    return;
  }
  JsonObject& root_0 = root[0];
  JsonObject& root_0_GeoPosition = root_0["GeoPosition"];
  float root_0_GeoPosition_Latitude = root_0_GeoPosition["Latitude"];
  float root_0_GeoPosition_Longitude = root_0_GeoPosition["Longitude"]; 
  Serial.println(root_0_GeoPosition_Latitude, 3); //Leh
  Serial.println(root_0_GeoPosition_Longitude, 3);
  client.stop();

ewaldc, 2019-10-28:
Assuming _client_ contains your JSON string, it's working perfectly on my system.
Can your print out _client_ before the parseArray statement ?
Somehow that must be corrupted or not null terminated.

Ewald

gabi68, 2019-10-28:
@ewaldc 
char servername1[] = "dataservice.accuweather.com";
void getlocation()                                //client function to send/receive GET request data.
{
  if (client.connect(servername1, 80))
  { //starts client connection, checks for connection
    client.println("GET /locations/v1/cities/search?apikey=4fUiwwSKqg48i2o1jGGHXmXggwzj6u8&q=Crawley%2C%20%2C%20West%20Sussex%2C%20UK HTTP/1.1");
    client.println("Host: dataservice.accuweather.com");
    client.println("User-Agent: ArduinoWiFi/1.1");
    client.println("Connection: close");
    client.println();
  }
  else {
    Serial.println("connection failed");        //error message if no client connect
    Serial.println();
  }
 
  const size_t capacity = JSON_ARRAY_SIZE(1) + JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(2) + 7 * JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(7) + JSON_OBJECT_SIZE(15) + 900;
  DynamicJsonBuffer jsonBuffer(capacity);
  //JsonObject& root = jsonBuffer.parseObject(client);
  Serial.println(client);
  JsonArray& root = jsonBuffer.parseArray(client);
  if (!root.success()) {
    Serial.println("Parsing failed!");
    return;
  }
  JsonObject& root_0 = root[0];
  JsonObject& root_0_GeoPosition = root_0["GeoPosition"];
  float root_0_GeoPosition_Latitude = root_0_GeoPosition["Latitude"];
  float root_0_GeoPosition_Longitude = root_0_GeoPosition["Longitude"];
  //Serial.println("Response:");
  Serial.println(root_0_GeoPosition_Latitude, 3); 
  Serial.println(root_0_GeoPosition_Longitude, 3);
  client.stop();
}
Can you post your working file? I must miss something.

Gabi

gabi68, 2019-10-29:
Still fighting with Parsing failed!. Anyone ...

ewaldc, 2019-10-29:
@gabi68,
working perfectly on an ESP8266 using your code.
PS don't share an API key publically...

```
#include <Arduino.h>
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
char servername1[] = "dataservice.accuweather.com";
void getlocation(); //client function to send/receive GET request data.
const size_t capacity = JSON_ARRAY_SIZE(1) + JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(2) + 7 * JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(7) + JSON_OBJECT_SIZE(15) + 900;
DynamicJsonBuffer jsonBuffer(capacity);
WiFiClient client;
String response;

void setup() {
  Serial.begin(115200);
  while (!Serial) {} // Wait for serial port initialization
  Serial.println();  //Clear some garbage that may be printed to the serial console
  WiFi.begin("myAP", "myPass");
  Serial.println("Connecting");
  while (WiFi.status() != WL_CONNECTED) delay(500);

  if (client.connect(servername1, 80))  { //starts client connection, checks for connection
      client.println("GET /locations/v1/cities/search?apikey=4fUiwwSKqg48i2o1jGGHXmXggwzj6u8&q=Crawley%2C%20%2C%20West%20Sussex%2C%20UK HTTP/1.1");
      client.println("Host: dataservice.accuweather.com");
      client.println("User-Agent: ArduinoWiFi/1.1");
      client.println("Connection: close");
      client.println();
    } else {
      Serial.println("connection failed"); //error message if no client connect
      Serial.println();
    }
    while (client.connected() || client.available()) {
      if (client.available())      {
        response = client.readStringUntil('\n');
        Serial.println(response);
      }
    }
  Serial.println(response);
  JsonArray& root = jsonBuffer.parseArray(response);
  if (!root.success()) {
    Serial.println("Parsing failed!");
    return;
  }
  JsonObject& root_0 = root[0];
  JsonObject& root_0_GeoPosition = root_0["GeoPosition"];
  float root_0_GeoPosition_Latitude = root_0_GeoPosition["Latitude"];
  float root_0_GeoPosition_Longitude = root_0_GeoPosition["Longitude"];
  //Serial.println("Response:");
  Serial.println(root_0_GeoPosition_Latitude, 3);
  Serial.println(root_0_GeoPosition_Longitude, 3);
  client.stop();
}
void loop() {
}
```

> Output
> HTTP/1.1 200 OK
> Date: Tue, 29 Oct 2019 13:45:50 GMT
> Content-Type: application/json; charset=utf-8
> Content-Length: 999
> Connection: close
> Server: Microsoft-IIS/10.0
> RateLimit-Limit: 50       
> RateLimit-Remaining: 45   
> Access-Control-Allow-Origin: *
> Access-Control-Allow-Methods: GET, PUT, POST, DELETE
> Access-Control-Allow-Headers: origin, x-requested-with, accept
> Access-Control-Max-Age: 3628800
> Cache-Control: public, max-age=85596
> Expires: Wed, 30 Oct 2019 13:32:26 GMT
> Host: api.accuweather.com
> User-Agent: ArduinoWiFi/1.1 
> X-Forwarded-For: 78.23.75.11
> X-Forwarded-Port: 80
> X-Forwarded-Proto: http
> 
> [{"Version":1,"Key":"715290","Type":"City","Rank":85,"LocalizedName":"Crawley","EnglishName":"Crawley","PrimaryPostalCode":"RH10 1","Region":{"ID":"EUR","LocalizedName":"Europe","EnglishName":"Europe"},"Country":{"ID":"GB","LocalizedName":"United Kingdom","EnglishName":"United Kingdom"},"AdministrativeArea":{"ID":"WSX","LocalizedName":"West Sussex","EnglishName":"West Sussex","Level":1,"LocalizedType":"Non-Metropolitan County","EnglishType":"Non-Metropolitan County","CountryID":"GB"},"TimeZone":{"Code":"GMT","Name":"Europe/London","GmtOffset":0.0,"IsDaylightSaving":false,"NextOffsetChange":"2020-03-29T01:00:00Z"},"GeoPosition":{"Latitude":51.113,"Longitude":-0.186,"Elevation":{"Metric":{"Value":76.0,"Unit":"m","UnitType":5},"Imperial":{"Value":249.0,"Unit":"ft","UnitType":0}}},"IsAlias":false,"SupplementalAdminAreas":[{"Level":0,"LocalizedName":"England","EnglishName":"England"},{"Level":2,"LocalizedName":"Crawley","EnglishName":"Crawley"}],"DataSets":["Alerts","MinuteCast","Radar"]}]
> [{"Version":1,"Key":"715290","Type":"City","Rank":85,"LocalizedName":"Crawley","EnglishName":"Crawley","PrimaryPostalCode":"RH10 1","Region":{"ID":"EUR","LocalizedName":"Europe","EnglishName":"Europe"},"Country":{"ID":"GB","LocalizedName":"United Kingdom","EnglishName":"United Kingdom"},"AdministrativeArea":{"ID":"WSX","LocalizedName":"West Sussex","EnglishName":"West Sussex","Level":1,"LocalizedType":"Non-Metropolitan County","EnglishType":"Non-Metropolitan County","CountryID":"GB"},"TimeZone":{"Code":"GMT","Name":"Europe/London","GmtOffset":0.0,"IsDaylightSaving":false,"NextOffsetChange":"2020-03-29T01:00:00Z"},"GeoPosition":{"Latitude":51.113,"Longitude":-0.186,"Elevation":{"Metric":{"Value":76.0,"Unit":"m","UnitType":5},"Imperial":{"Value":249.0,"Unit":"ft","UnitType":0}}},"IsAlias":false,"SupplementalAdminAreas":[{"Level":0,"LocalizedName":"England","EnglishName":"England"},{"Level":2,"LocalizedName":"Crawley","EnglishName":"Crawley"}],"DataSets":["Alerts","MinuteCast","Radar"]}]
> 51.113
> -0.186

gabi68, 2019-11-01:
@ewaldc ,

Many thanks.

Gabi

closed 2019-11-21 by bblanchon