bblanchon/ArduinoJson #1117: TextFormatter.hpp request for member write in ArduinoJson12  _length += _writer.write reinterpret cast

quique123, 2019-10-27:
I'm trying to use ArduinoJson with Ken Taylor's Autoconnect library for an esp01.  I think I successfully migrated from AJ5 to AJ6 as explained here:

https://forum.arduino.cc/index.php?topic=643132.0

But I get this error.  Did I do something wrong or is it something else?

quique123, 2019-10-28:
I got it working but now i have the esp01s disconnect after EXIT PORTAL button click, but not be able to connect again with result 0.

https://forum.arduino.cc/index.php?topic=643132.msg4350862#msg4350862

bblanchon, 2019-10-29:
Hi @quique123,

I'm afraid I cannot help you because I don't have experience with this project.

Do you still have a question regarding ArduinoJson?

Best Regards,
Benoit

closed 2019-11-26 by bblanchon