bblanchon/ArduinoJson #1119: Memory leak porting from V5 to V6 - what am I doing wrong?

ewaldc, 2019-10-28:
While porting from V5 (_v5.11.5_ & _V5.13.4_ tested) to V6 (_V6.12.0_), seamingly similar code is leaking memory: 504 bytes, about the size of the JSON document which is 497 bytes in V6.  
Interestingly, for _V5.13.4_ the assistant has this string as 587 bytes, but it's reported as 616 bytes by _size()_.   
Please note that V5 is also leaking memory, unless I issue both _jsonDoc->clear();_ and _delete jsonDoc;_  otherwise about 24 bytes are leaked.   

The full code is running on V5.11.5 24/7 for over a year now, parsing ~500 JSON strings a day without issue, some of which need nesting levels of 20+ and are just ~5K smaller than the ESP available memory .  No other Json library even came close with respect to memory footprint, performance and stability !  V6 promises to even use less memory, hence the port.

The MVCE code compiles both with 5.13.4 and 6.12.0 (using some quick & dirty tricks to minimize code size and #ifdefs)
In essence the code parses a JSON string (customer-supplied, received over HTTPS and of unknown size) and solves the problem of determining the optimum & minimal size for the DynamicJsonDocument by doing a two-pass deserialization. The first pass uses the available ESP memory (minus 4K), the seconds one uses the exact size as determined by _memoryUsage()_.  Hence it violates the best practice of using a JsonDocument only ones, but I have not found a better way to deal with JSON strings of unknown length...

What I am doing wrong?
**Thanks in advance**, 
Ewald

* target platform: _Espressif 8266 2.2.3 > NodeMCU 1.0 (ESP-12E Module) - ESP8266 Arduino 2.5.2_
* compiler model and version: _platformio 1.9_
* MVCE 
```
#include <ArduinoJson.h>
#include <Arduino.h>

template <typename T, typename U> static inline void DbgPrint(T x, U y) {Serial.print(x); Serial.print(y);}
template <typename T, typename U> static inline void DbgPrintln(T x, U y) {Serial.print(x); Serial.println(y);}
template <typename T> static inline void DbgPrintln(T x) {Serial.println(x);}
template <typename T> static inline void DbgPrint(T x) {Serial.print(x);}

#if ARDUINOJSON_VERSION_MAJOR == 5
#define memoryUsage size
#define DynamicJsonDocument DynamicJsonBuffer
#define DeserializationError boolean
JsonVariant jsonObj;
#endif

const char* json = "{\"status\":\"CREATED\",\"context\":{\"headers\":{\"Location\":[\"/rest/sitemaps/events/XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX\"]},\"committingOutputStream\":{\"bufferSize\":0,\"directWrite\":true,\"isCommitted\":false,\"isClosed\":false},\"entityAnnotations\":[],\"entityStream\":{\"bufferSize\":0,\"directWrite\":true,\"isCommitted\":false,\"isClosed\":false}}}";
static int count = 0;

DynamicJsonDocument* getJsonDoc() {
	DeserializationError error;
	boolean final = false;
	size_t sz = ESP.getFreeHeap() - 4096; //Free heap space - 4K
	do {
#if ARDUINOJSON_VERSION_MAJOR == 6
		DynamicJsonDocument *jsonDoc = new DynamicJsonDocument(sz);
		error = deserializeJson(*jsonDoc, json);
		if (error != DeserializationError::Ok) goto errout;
#else
		DynamicJsonDocument *jsonDoc = new DynamicJsonDocument(sz);
		jsonObj = jsonDoc->parseObject(json);
		if (jsonObj != jsonObj.success()) goto errout;
#endif
		else {
			if (final) return jsonDoc;
			final = true;
			sz = jsonDoc->memoryUsage();
			DbgPrintln(F(" - final jsondoc size: "), sz);
		}
		jsonDoc->clear();
		delete jsonDoc;
		DbgPrintln(F("free heap memory after delete: "), ESP.getFreeHeap());
	} while (!error || !final);
errout:
	DbgPrintln(F(" - deserializeJson failed: "));  //, error.c_str()
	return nullptr;
}

ICACHE_FLASH_ATTR void handleSubscribe(boolean useJsonString = false) {
	DbgPrintln(F("free heap memory @start: "), ESP.getFreeHeap());
  	DynamicJsonDocument* jsonDoc;
	jsonDoc = getJsonDoc();
	if (jsonDoc) {
#if ARDUINOJSON_VERSION_MAJOR == 6
	serializeJson(*jsonDoc, Serial);
#else
	jsonObj.printTo(Serial);
#endif		
		Serial.println();
   		jsonDoc->clear();
   		free(jsonDoc);
	}
	DbgPrintln(F("free heap memory @exit: "), ESP.getFreeHeap());
}

void setup(void) {
  Serial.begin(115200);
  while (!Serial) {} // Wait for serial port initialization
  Serial.println();  //Clear some garbage that may be printed to the serial console
  Serial.println(ARDUINOJSON_VERSION_MAJOR);
}
void loop() {
  while (count++ < 2) {
    handleSubscribe(false);
	handleSubscribe(true);
  }
}
```

bblanchon, 2019-10-29:
Hi @ewaldc,

I think the problem is the call to `free(jsonDoc)` instead of `delete jsonDoc`.

Best regards,
Benoit

ewaldc, 2019-10-29:
Yes!  I had changed it to _free_ since the compiler gave errors on V5 (complaining there was no destructor function).
So _jsonDoc->clear();_ does not release the memory ?
From the API doc:

> JsonDocument::clear() clears the JsonDocument and releases all the memory from the memory pool.
> 
> Unlike JsonArray::clear() and JsonObject::clear(), this function releases the memory from the memory pool; therefore, it doesn’t cause a memory leak.

Anyhow, problem solved!

bblanchon, 2019-11-01:
Hi @ewaldc,

Just because `clear()` doesn't fix your issue doesn't mean it does nothing at all :-)

If I had to implement two-pass parsing with ArduinoJson 6, I would write this:

```c++
// Returns the maximum acceptable capacity
size_t maxAllowedCapacity() {
  return ESP.getFreeHeap() - 4096;
}

// Returns the capacity required to deserialize the specified JSON document,
// or returns 0 if deserialization fails
size_t measureCapacity(const char* json) {
  DynamicJsonDocument doc(maxAllowedCapacity());
  return deserializeJson(doc, json) ? 0 : doc.memoryUsage();
}

// Deserializes the JSON document in two passes
// If deserialization fails, returns a "null" JsonDocument (.isNull() == true)
DynamicJsonDocument deserializeJsonTwoPasses(const char* json) {
  DynamicJsonDocument doc(measureCapacity(json));
  deserializeJson(doc, json);
  return doc;
}
```

I didn't test this code, but I'm quite confident it should work.

Anyway, it's a pity you have to do two-pass deserialization like that; I'll try to add `DynamicJsonDocument::shrinkToFit()` in version 6.14.

Best Regards,
Benoit

ewaldc, 2019-11-01:
Hi Benoit,
`DynamicJsonDocument::shrinkToFit()` would be great !

All works fine except for the last function in the suggested code.  I guess that when  _deserializeJsonTwoPasses_ returns a DynamicJsonDocument, it temporarily requires 2x the size of the document, mostly resulting in a null document when the JSON string is large.   But this works:
```
DynamicJsonDocument* getJsonDocFromFile(File f) {
  DynamicJsonDocument* doc = new DynamicJsonDocument(measureCapacity(f));
  f.seek(0); // make sure file pointer is at beginning of file
  deserializeJson(*doc, f, DeserializationOption::NestingLimit(20));
  return doc;
}
```

The code is now more readable and compact, and more RAM is now available.
Warm regards,
Ewald

bblanchon, 2019-11-02:
Hi Ewald,

> when deserializeJsonTwoPasses returns a DynamicJsonDocument, it temporarily requires 2x the size of the document

It should not because return value optimization (RVO) applies.
I tested the following code with GCC 4.9.3, and I can confirm that RVO works.

```c++
DynamicJsonDocument doc = deserializeJsonTwoPasses(json);
```

I guess that there is something else in your program that causes the unwanted copy.

If you need to track the allocations of `DynamicJsonDocument`, add log in `DefaultAllocator::allocate()` and `DefaultAllocator::free()`.
Alternatively, you can also use a [custom allocator class](https://arduinojson.org/v6/api/basicjsondocument/).

Best regards,
Benoit

ewaldc, 2019-11-02:
Hi Benoit,
You are right (again)!  After upgrading the _xtensa-lx106-elf-gcc_ compiler to the latest version, enabling c++version 11 and changing the way I initialize my classes (passing  _deserializeJsonTwoPasses(json)_ directly to the constructor), all of which were needed, the double copy problem has disappeared.  Saved about 880 bytes of memory for a 40K Json string compared to V5.x code.   The two pass deserialization might sound a bit far fetched, but it's doing the job well and fast (about 1 second). 
Thanks so much!

closed 2019-11-02 by ewaldc

bblanchon, 2019-11-07:
Hi @ewaldc,

I added `shrinkToFit()` on the "6.x" branch. It simplifies your code to:

```c++
DynamicJsonDocument deserializeJson(const char* json) {
  DynamicJsonDocument doc(ESP.getFreeHeap() - 4096;);
  deserializeJson(doc, json);
  doc.shrinkToFit();
  return doc;
}
```

Can you give it a try?

Best Regards,
Benoit

ewaldc, 2019-11-08:
Hello Benoit,
Works to **perfection** as measured by a 12 hour stress test (test continues for another 84 hours...)
I put this into my "platformio.ini" file, which resolved to _<ArduinoJson> 6.13.0 #062c1c1_
Hope that was what you meant...

```
lib_deps =
;  ArduinoJson@6.11.5
;  ArduinoJson@6.13.0
    https://github.com/bblanchon/ArduinoJson#6.x
```

Thanks so much,
Ewald