bblanchon/ArduinoJson #1120: Using auto type to access array nested in object crashes device

prototypicalpro, 2019-10-28:
**Description**
Running the following code snippet works great:
```C++
StaticJsonDocument<500> doc;
constexpr char str[] = "{\"contents\":[{\"module\":\"Packet\"},{\"module\":\"Analog\"}]}";
deserializeJson(doc, str);
JsonObject sensor = thing["contents"][1];
Serial.println(sensor.isNull()); // <- prints "0"
```
however, changing the type to `auto` causes my device to hang:
```C++
StaticJsonDocument<500> doc;
constexpr char str[] = "{\"contents\":[{\"module\":\"Packet\"},{\"module\":\"Analog\"}]}";
deserializeJson(doc, str);
auto sensor = thing["contents"][1];
Serial.println(sensor.isNull()); // <- Never printed, and must be reset to continue
```
I suspect this issue is due to `ElementProxy` type not resolving properly, but am unsure why this would be causing the device to crash.

**Platform**
I am using ArduinoJson 6.12.0, Arduino IDE 1.8.10, and running this sketch on the Adafruit Feather M0 (SAMD21) with Adafruit SAMD Core 1.5.2 . I've attached the
[relevant sketch](https://github.com/bblanchon/ArduinoJson/files/3780245/jsontest.txt), as well as the [build log.](https://github.com/bblanchon/ArduinoJson/files/3780253/build.txt)

ewaldc, 2019-10-29:
@prototypicalpro,
```
auto sensor = thing["contents"][1];
decltype(sensor) b;
```
src\main.cpp:15:20: error: no matching function for call to 'ArduinoJson6120_000001::ElementProxy<const ArduinoJson6120_000001::MemberProxy<const ArduinoJson6120_000001::ObjectRef&, const char*>&>::ElementProxy()'
   decltype(sensor) b;

So indeed it resolves to class _ElementProxy_, which overlays the = operator, so the problem could be there somewhere.

PS the code also crashes/resets an ESP8266 (and other devices)

closed 2019-10-30 by bblanchon

bblanchon, 2019-11-01:
Fix published in [ArduinoJson 6.13.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.13.0)

commit 6da6f921 (2019-10-30): Fixed dangling reference in MemberProxy and ElementProxy (fixes #1120)