bblanchon/ArduinoJson #1121: Appending to a JSON file...

ukadiyala, 2019-10-28:
Hi There,

You have done a fantastic job of writing this library and keeping it up to date. Thank you for all your efforts.

I'm trying to utilise your library on an ESP8266 dev board. Utilising vscode & platformio with Arduino libraries. The idea is to be able to add data to a JSON file on SPIFFS.

1. I need to be able to add to a file structured similar to the following:
{"ENTRY":{"TYPE":"EVENT", "DESC":"INIT..."}, "ENTRY":{"TYPE":"EVENT", "DESC":"REBOOT..."}, "ENTRY":{"TYPE":"EVENT", "DESC":"GPIO 1 UP"}}

2. Here is a snippet where I'm trying to add to the file:
---
JsonObject& root = jBuff.parseObject(fileData);
JsonObject& newEvent = root.createNestedObject("ENTRY");
if(!newEvent.success())
{
    Serial.println("Unable to create JSON element. Memory allocation failed...");
}
else
{
    newEvent["TYPE"] = "EVENT"; 
    newEvent["DESC"] = "GPIO 1 UP";
}

String opslogData = "";
root.printTo(opslogData);
---
This results in the contents of the log file being overwritten and yeilds the following result:
{"ENTRY":{"TYPE":"EVENT", "DESC":"GPIO 1 UP"}}

3. I have tried this in both V5 and V6 API formats with the same result. I was under the impression that createNestedObject adds to the existing file. What am I missing? 

Please help. Thank you for your time.

Regards,
Udhay

bblanchon, 2019-10-30:
Hi @ukadiyala,

This is not a valid JSON document:

```json
{
  "ENTRY":{"TYPE":"EVENT", "DESC":"INIT..."},
  "ENTRY":{"TYPE":"EVENT", "DESC":"REBOOT..."},
  "ENTRY":{"TYPE":"EVENT", "DESC":"GPIO 1 UP"}
}
```

Indeed, you cannot have the same key repeated in one object.

I recommend you use this schema instead:

```json
{
  "ENTRIES": [
    {"TYPE":"EVENT", "DESC":"INIT..."},
    {"TYPE":"EVENT", "DESC":"REBOOT..."},
    {"TYPE":"EVENT", "DESC":"GPIO 1 UP"}
  ]
}
```

Alternatively, you can use the [JSON Lines](http://jsonlines.org/) format:

```json
{"TYPE":"EVENT", "DESC":"INIT..."}
{"TYPE":"EVENT", "DESC":"REBOOT..."}
{"TYPE":"EVENT", "DESC":"GPIO 1 UP"}
```

Best Regards,
Benoit

ukadiyala, 2019-10-30:
Thank you Benoit.
You are correct. I was about to post admitting the same. That was a stupid question. Apologies.

It has been a long time since I got back to the ESP. I'm trying to following without much success.
Was going through the Arduino reference to refresh my mind. 

    JsonArray *events;
    JsonObject *root;

    if(fileData=="null")
    {
        Serial.println("Unable to add device log information to " + filePath);        
    }
    else
    {
        root = jBuff.parseObject(fileData);
        
        if(!root.success())
        {
            root = jBuff.createObject();
            events = root.createNestedArray("EVENTS");
        }
        else
        {
            events = root["EVENTS"];
        }

        for (size_t i = 0; i < 5; i++)
        {
            JsonObject& new_event = events.createNestedObject();
            new_event["TITLE"] = "TEST EVENT";
            new_event["DESC"] = "TESTING " + (char)i; 
        }
        
        String new_event_group = "";
        root.printTo(new_event_group);
    }

Any suggestions?

bblanchon, 2019-11-21:
You cannot write `root = jBuff.parseObject(fileData);` with ArduinoJson 5 because [parseObject()](https://arduinojson.org/v5/api/jsonbuffer/parseobject/) returns a reference.
I really recommend to [upgrade to ArduinoJson 6](https://arduinojson.org/v6/doc/upgrade/) which doesn't have this problem.

closed 2019-12-09 by bblanchon