bblanchon/ArduinoJson #1123: Send JSON through HTTP POST Request

isosel, 2019-10-31:
Hello Benoit,

i'm working on the arduino MKR Wifi 1010 and my goal is to send a HTTP POST request with a JSON object to a connected device in order to switch it on or off. This device is connected to my network as my arduino board. Once both are connected to my network I send this HTTP request: 

WiFiClient myClient1;

    postData = "{\"header\":{\"from\":\"/app/339299-e1853779db9fc7866cf6dfe0b697b0b8/subscribe\",\"messageId\":\"36f3d4b5da1545dd127176bdbbf8f2cf\",\"method\":\"SET\",\"namespace\":\"Appliance.Control.ToggleX\",\"payloadVersion\":1,\"sign\":\"195a5ea4fb950c3cb475cd5daa5d4e64\",\"timestamp\":1571919319},\"payload\":{\"togglex\":{\"channel\":0,\"onoff\":1}}}";
    
  if (myClient1.connect("192.168.1.132, 80)) 
  {
    Serial.println("Connected.");
    //HTTP request 1st version:
    myClient1.println("POST /config HTTP/1.1");
    myClient1.println("Host:192.168.1.132");
    myClient1.println("Content-Type:application/json; charset=UTF-8");
    myClient1.println("Connection:close");
    myClient1.print("Content-Length:");myClient1.println(postData.length());//208
    myClient1.println("Accept-Encoding:gzip");
    myClient1.println("User-Agent:ARDUINO1");
    myClient1.println();
    myClient1.println(postData);
    Serial.print("Code received:");
    Serial.println(myClient1.read());
  }
  else
  {
    Serial.println("Connection failed.");
  } 

When I use Curl or Postman, there is no problem, it works very well but when I am using Arduino, it seems something is wrong but I don't know what.

[Link to my github project](https://github.com/isosel/meross-mkr1010/wiki/%234-Postman-configuration)

When I discovered your library, I thought hmmm maybe it's because I send the JSON object in a String form wich is not correct, and Postman or Curl transform the JSON String object itself into a real JSON Object, i don't know...

My questions:

In your example : "https://github.com/bblanchon/ArduinoJson/blob/6.x/examples/StringExample/StringExample.ino"

I can deserialize my postData string and replace myClient1.println(postData); by myClient1.println(postDataJSON);

using this code:

  DynamicJsonDocument doc(postData.length()); // doc(208) because this is the length of my postData string.
  String input = postData;
  deserializeJson(doc, input);
  JsonObject postDataJSON = doc.as<JsonObject>();

Do you have a suggestion ? I found no answer on github/internet/forums and I am suffering xD

Best regards,

isoSeL.

bblanchon, 2019-11-01:
Hi isoSeL,

Before looking at ArduinoJson, we must understand why the hard-coded version doesn't work.
I carefully looked at the code above, but I could not spot the error.

I suggest that you compare the message sent by the Arduino with the one from cURL.
To do that, I recommend using the `LoggingClient` from the [StreamUtils library](https://github.com/bblanchon/ArduinoStreamUtils):

```c++
LoggingClient loggingClient(myClient1, Serial);
// then replace every occurence of "myClient1" with "loggingClient"
```

`LoggingClient` will log everything you send to the server, so you can easily compare it with cURL.

Good luck!

Best Regards,
Benoit

ewaldc, 2019-11-02:
@isosel,
Most likely the problem is here:
`DynamicJsonDocument doc(postData.length()); // doc(208) because this is the length of my postData string.`
The JsonDocument needs more space than the size of the Json string. You can check with the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) how much is needed.
When the strings arrive in HTTP(S) POST's and might differ in size, the challenge is of course that you can't leverage the assistant and you will have to make a guess on the size that is needed.  This can be done by adding a solid amount of extra space (e.g. take 2x the size) or by leveraging a _two pass deserialization_ that Benoit helped me to write and optimize [here](https://github.com/bblanchon/ArduinoJson/issues/1119). 
Good luck with your project.
Ewald

closed 2019-11-21 by bblanchon