bblanchon/ArduinoJson #1124: Using std::max() to define StaticJsonDocument<>

lbussy, 2019-11-03:
Hello Benoit,

I have a section of my code I am trying to re-use and it deals with sending the same data to different endpoints and those endpoints require a specific format.  To that end, the capacity is different for each.

My best idea to help make the code somewhat easy to remember (and define the doc outside some conditional formatting) is using std::max():

```
const size_t capacity = max(
    JSON_OBJECT_SIZE(8) + 210, // BF or Target
    JSON_ARRAY_SIZE(1) + JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(8) + 25 // Thingspeak
);
StaticJsonDocument<capacity> doc;

if (condition) {
    // Create JSON one way
} else {
    // Create JSON a different way
}
// Create connection and POST results
```

When I try this I get an error: "expression must have a constant value -- the value of variable "capacity" (declared at line 131) cannot be used as a constant".  I understand the words of course, that when there's a "choice" that mutability is carried forward even though I am declaring capacity as const.  I just don't understand why.

This also may not be the best way to deal with it so maybe there is a better way to do this??

Regards,

- LB

bblanchon, 2019-11-03:
Hi @lbussy,

Try to replace the first `const` with `constexpr` to force evaluation at compile-time.

Then, if you do use [`std::max()`](https://en.cppreference.com/w/cpp/algorithm/max) (and not Arduino's `max()` macro), you'll need to compile with C++14, because [`std::max()`](https://en.cppreference.com/w/cpp/algorithm/max) was not declared `constexpr` in C++11.

Best Regards,
Benoit

lbussy, 2019-11-03:
Thank you for the reply Benoit.

Strangely, even if I included Arduino.h the expression was still being handled by std::max().  I ended up putting the following macros in my header:

```
#define max(a,b) ((a)>(b)?(a):(b))
#define min(a,b) ((a)<(b)?(a):(b))
```

Using `constexpr` was the ticket though, thank you for the help there.  I understand that better.

I did find a (closed) [issue on the Arduino repo](https://github.com/esp8266/Arduino/issues/398), which comes close to describing this and led me to using the macros.  That issue suggests I am passing in two different types, however even when I explicitly cast the valies to be size_t it fails.

I have it fixed, so I guess I'm happy either way.

bblanchon, 2019-11-04:
Cool. I'm glad you fixed your issue.
Thank you for using ArduinoJson and good luck with your projects.

closed 2019-11-04 by bblanchon