bblanchon/ArduinoJson #1126: 6.13.0 broken for non-copy-constructable String types

mikee47, 2019-11-05:
With commit 6da6f921cd7667296f64ee4c641f61df13d24727 both ElementProxy and MemberProxy take copies of the
TString argument instead of taking a reference.

This breaks existing code which depends upon the referencing behaviour, and also has performance/efficiency implications.

The fault addressed by the above commit is briefly outlined in #1120, caused by a `dangling reference`. To clarify, this is where the ElementProxy/MemberProxy take a reference to a temporary object (e.g. arduino String) which the compiler optimises out before the reference is used. This leads to unpredictable behaviour, so clearly taking proper copies is the safest (default) behaviour.

However, I have a custom `FlashString` class (with adapter) which is not copy-constructible and so now no longer works as intended.

Here are the details:

* [FlashString](https://github.com/SmingHub/Sming/blob/develop/Sming/Wiring/FlashString.h)
* ArduinoJson [FlashStringRefAdapter](https://github.com/SmingHub/Sming/blob/develop/Sming/Libraries/ArduinoJson6/include/FlashStringRefAdapter.hpp)

I've been unable to find a workable solution without modifying ArduinoJson so I'll propose that and see where we go.

It feels like there's room for improvement here. Can we detect when it's safe to take a reference, and do that when possible?

bblanchon, 2019-11-05:
Hi @mikee47,

I'm sorry I break your code; I didn't see that coming.
As you understood, I had to make this change to fix #1120, and I believe it was the right fix.

Mike, why is the `FlashString` non-copiable?

Best Regards,
Benoit

closed 2019-11-09 by mikee47

mikee47, 2019-11-20:
@bblanchon Just to let you know I've fixed the incompatibility problem by adding copy constructor support. Inspired by your own libraries, I've created [FlashString](https://github.com/mikee47/FlashString/tree/develop). Thank you for setting the standard!