bblanchon/ArduinoJson #1128: Possible to serialize static objects

m-andersen, 2019-11-07:
Hi

In my project I have different JSON documents that are generated and send to a browser. But sometimes the response is always the same (static JSON content).
Is it possible to create a const JSON document/object that can be serialized wihtout adding the key/values everytime to a document?

Further, I read in the documentation that if another JsonObject is assigned to a document, it is always done by copying/cloning the whole object.
Is it possible to specify that this other object is const so no copy is needed? The Json document could reference the static JsonObject I mentioned before without copy.

Thanks

ewaldc, 2019-11-08:
Hi @m-andersen,
I'm using the ESP platform and keep the static Json Strings in flash either as const char * or as a file (e.g. via SPIFFS, see code below).  If you return the JsonDocument from a function, the compiler will optimize out the extra copy, so there is no need to work with pointers to JsonDocument.
     
You can use a linked list/vector to keep  track of all pre-parsed JSON strings.  You could lookup teh JSON strings by name (see code) or eventually by hash string (after JSON minimise of course) from text (e.g. if you want to check if a JSON string supplied over the web matches a cached one).
The biggest challenge is to determine the size of the JsonDocument for each string.  One option is to lookup the needed size for each static string using the ArduinoJson assistant and store that somewhere or use a two pass method as in the code below (in my case the JSON strings come over the web, so I can never know the size in advance).

```
size_t measureCapacity(File f) {
  DynamicJsonDocument doc(ESP.getFreeHeap() - 4096);
  return deserializeJson(doc, f, DeserializationOption::NestingLimit(20)) ? 0 : doc.memoryUsage();
}

// Deserializes the JSON document in two passes
// If deserialization fails, returns a "null" JsonDocument (.isNull() == true)
DynamicJsonDocument getJsonDocFromFile(File f) {
  DynamicJsonDocument doc(measureCapacity(f));
  f.seek(0); // make sure file pointer is at beginning of file
  deserializeJson(doc, f, DeserializationOption::NestingLimit(20));
  return doc;
}

struct ParsedJsonMap {
	const char *name;
	DynamicJsonDocument jsonDoc;
	ParsedJsonMap *next;
};

ParsedJsonMap *parsedJsonMapList = new ParsedJsonMap {name, getJsonDocFromFile(f), parsedJsonMapList}; 
```

Hope this helps.
Ewald

bblanchon, 2019-11-08:
Hi @m-andersen,

For static segments of JSON, you can wrap a string with [serialized()](https://arduinojson.org/v6/api/misc/serialized/):

```c++
doc["key"] = serialized("[1,2,3]");
```

This allows adding a static part in a `JsonDocument`.

Of course, if the whole JSON document is static, then you can simply send a static string in the response; you don't need ArduinoJson for that.

To answer your second question, no, it's not possible to disable the copy. ArduinoJson 5 worked in the exact opposite; it always stored pointers. I decided to copy members in ArduinoJson 6 because it allows returning `JsonDocument` without worrying about dangling pointers.

However, if there is a legitimate use case, I can extend the API to explicitly store a reference.
Can you describe your use case?

Best Regards,
Benoit

closed 2019-11-21 by bblanchon