bblanchon/ArduinoJson #1130: 新的BUG问题

yunxiang521, 2019-11-12:
你好，我在使用ArduinoJson测试Json格式字符串是否完整情况下，出现了系统崩溃的情况。
当Json格式字符串："{"ss":s}"，失去前一个”{“符号时， 在使用deserializeJson转换时整个系统会崩溃。
例如把Json字符串格式 ”ss“:s}进行转换。得到的是如下崩溃错误。
Exception (28):
epc1=0x4020b2cb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00005a53 depc=0x00000000

我的芯片是ESP8266。

bblanchon, 2019-11-14:
你好，

请编写示例程序。
如果可能，请用英文写，我听不懂中文，该文字来自Google翻译。

最好的祝福，
Benoit

yunxiang521, 2019-11-19:
Hello, when I used arduinojson to test whether the JSON format string is complete, the system crashed.
When the JSON format string: "{" s s ": s}", loses the previous "{" symbol, the whole system will crash when using deserialize JSON conversion.
For example, convert the JSON string format "s s": s}. What you get is the following crash error.
Exception (28):
epc1=0x4020b2cb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00005a53 depc=0x00000000
My chip is esp8266.

yunxiang521, 2019-11-19:
My English is also very poor. I can only rely on Internet translation,
Please test whether the following string JSON is correct.
"SDF": "123"}

bblanchon, 2019-11-19:
Hi @yunxiang521,

Please show the source code.
Also, you may use [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) to get more information about the crash.

Best Regards,
Benoit

closed 2019-12-09 by bblanchon