bblanchon/ArduinoJson #1131: serializeJson(doc1,http) error

BLimaM4I, 2019-11-13:
Hello!

I have a simple project where the line of code
 "serializeJson(doc1,http);" gives me errors. I have also tried putting serializeJson(doc1, (Client&)http) but then the ESP32 keeps reseting with brownout error. I have also put a capacitor between 5V and GND but still the same error; 

Error when using only "serializeJson(doc1,http);:

Processing nodemcu-32s (platform: espressif32; board: nodemcu-32s; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 1.11.1 > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.12.0
|-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <NTPClient> 3.1.0
|-- <WiFi> 1.0
Building in release mode
Compiling .pio\build\nodemcu-32s\src\main.cpp.o
In file included from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.hpp:34:0,
                 from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.h:9,
                 from src\main.cpp:9:
C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp: In instantiation of 'size_t ArduinoJson6120_000001::serializeJson(const TSource&, TDestination&) [with TSource = ArduinoJson6120_000001::StaticJsonDocument<112u>; TDestination = HTTPClient; size_t = unsigned int]':
src\main.cpp:135:28:   required from here
C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:102:35: error: no matching function for call to 'serialize(const ArduinoJson6120_000001::StaticJsonDocument<112u>&, HTTPClient&)'
   return serialize<JsonSerializer>(source, destination);
                                   ^
In file included from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:10:0,
                 from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.hpp:34,
                 from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.h:9,
                 from src\main.cpp:9:
C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:34:8: note: candidate: template<template<class> class TSerializer, class TSource> size_t ArduinoJson6120_000001::serialize(const TSource&, Print&)
 size_t serialize(const TSource &source, Print &destination) {
        ^
C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:34:8: note:   template argument deduction/substitution failed:
In file included from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.hpp:34:0,
                 from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.h:9,
                 from src\main.cpp:9:
C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:102:35: note:   cannot convert 'destination' (type 'HTTPClient') to type 'Print&' 
   return serialize<JsonSerializer>(source, destination);
                                   ^
In file included from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:10:0,
                 from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.hpp:34,
                 from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.h:9,
                 from src\main.cpp:9:
C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:40:8: note: candidate: template<template<class> class TSerializer, class TSource> size_t ArduinoJson6120_000001::serialize(const TSource&, char*, size_t)
 size_t serialize(const TSource &source, char *buffer, size_t bufferSize) {
        ^
C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:40:8: note:   template argument deduction/substitution failed:
In file included from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.hpp:34:0,
                 from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.h:9,
                 from src\main.cpp:9:
C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:102:35: note:   cannot convert 'destination' (type 'HTTPClient') to type 'char*'  
   return serialize<JsonSerializer>(source, destination);
                                   ^
In file included from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:10:0,
                 from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.hpp:34,
                 from C:\users\bruno\.platformio\lib\ArduinoJson\src/ArduinoJson.h:9,
                 from src\main.cpp:9:
[... 135 lines cut ...]
    // Extract time
    int splitT = formattedDate.indexOf("T");
    timeStamp = formattedDate.substring(splitT + 1, formattedDate.length() - 1);

    //Start filling in the JSON Objects
    doc1["gateway_id"] = "ESP_01";
    doc1["sensor_id"] = "0101";
    doc1["Sensor1"] = av1;
    doc1["Sensor2"] = av2;
    doc1["Sensor3"] = av3;
    doc1["Sensor4"] = av4;
    doc1["time"] = timeStamp;
       
    //Start the request by HTTP to the server
    HTTPClient http;
    http.begin("http://??????????????????????/measurements");
    http.addHeader("Content-Type", "application/json");
    serializeJson(doc1,Serial);
    serializeJson(doc1,http);
    //Close the HTTP connection
    http.end();
    // Extract time
    Serial.println(timeStamp);
    delay(2000);
  }

bblanchon, 2019-11-14:
Hi @BLimaM4I,

Here are the relevant parts of the compiler output:

```
error: no matching function for call to 'serialize(const StaticJsonDocument<112u>&, HTTPClient&)'
note: candidate: template<template<class> class TSerializer, class TSource> size_t serialize(const TSource&, Print&)
note:   cannot convert 'destination' (type 'HTTPClient') to type 'Print&'
```

As you can see, the compiler cannot convert `HTTPClient` to `Print&` (the "output stream" class of Arduino). In other words, `HTTPClient` doesn't implement the `Print` interface; that's why no overload of `serializeJson()` matches, and that's why the ESP crashes when you force the conversion to `Client&`.

I don't know where the `HTTPClient` class comes from, but it doesn't implement the streaming interfaces as you expect. You can try to upgrade to [ArduinoJson 6.13.0](https://arduinojson.org/2019/11/01/version-6-13-0/), which supports custom writer classes or use another HTTP implementation.

Best regards,
Benoit

ewaldc, 2019-11-14:
@BLimaM4I,
WiFiClient class is derived from Stream and thus can be used as a target for _serializeJson_ and
can also be used with HTTPClient (and/or with ESP8266[Async]WebServer)
Examples [here](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266HTTPClient/examples)
Use as follows:

```
WiFiClient client;
HTTPClient http;
if (http.begin(client, "http://wwwmysite.org/")) {  // HTTP
```

BLimaM4I, 2019-11-15:
It worked!
Thanks for the high quality inputs!

closed 2019-11-16 by bblanchon