bblanchon/ArduinoJson #1132: Deserialization do not work with Serial.read

mois770, 2019-11-17:
Hi,
I can't figure out why this doesn't work.

I made a little change on the deserialization example (Quickstart). Instead of having the char json[] inside the program, I wanted to do it with Serial.read, like the code below.
If I insert the value of char[] through Serial it returns "Incomplete Input" and "Invalid Input". But if I uncomment the value of char [] it gives the right JSON. I know that I used the same value of char [] because the function showNewData shows the data in.
-------
const byte numChars = 32;
char receivedChars[numChars];  // an array to store the received data

void setup() {
 Serial.begin(115200);
 Serial.println("Started,");
}
void loop() {
  recvWithEndMarker();
  showNewData();
}
void recvWithEndMarker() {
  static byte ndx = 0;
  char endMarker = '\n';
  char rc;
  while (Serial.available() > 0 && newData == false) {
     rc = Serial.read();
    if (rc != endMarker) {
      receivedChars[ndx] = rc;
      ndx++;
      if (ndx >= numChars) {
        ndx = numChars - 1;
      }
    }
    else {
      receivedChars[ndx] = '\0'; // terminate the string
      ndx = 0;
      newData = true;
    }
// char receivedChars[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

 StaticJsonDocument<200> doc;
 DeserializationError error = deserializeJson(doc,receivedChars);
 if (error) {
  Serial.print(F("deserializeJson() failed: "));
  Serial.println(error.c_str());
  return;
 }
 const char* sensor = doc["sensor"];
 long time = doc["time"];
 double latitude = doc["data"][0];
 double longitude = doc["data"][1];
 // Print values.
 Serial.println(sensor);
 Serial.println(time);
 Serial.println(latitude, 6);
 Serial.println(longitude, 6);
  }

void showNewData() {
    if (newData == true) {
        Serial.print("This just in ... ");
        Serial.println(receivedChars);// show the incoming data
        newData = false;
    }   
}

bblanchon, 2019-11-18:
Hi @mois770,

32 bytes are not enough to store the input string; it requires 64 bytes at least.

BTW, your life would be much easier if you passed `Serial` directly to [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/):

```c++
DeserializationError error = deserializeJson(doc, Serial);
```

Best regards,
Benoit

mois770, 2019-11-18:
Hi, Benoit
Thank you for the answer.
I saw the possibility you mentioned. But I was not succeed to implement.
Also, I pretend to send JSON with bluetooth, that uses SerialBT instead of Serial.
But if it's possible to use with SerialBT, I would ask you some example.
And I already wrote that I found your answer on issue #1121, and it worked!
Best regards

bblanchon, 2019-11-19:
Hi @mois770,

I never used `BluetoothSerial`, but since the class derives from `Stream`, it should work with ArduinoJson.
Maybe, @me-no-dev, who is a long-time user of ArduinoJson on ESP, could cast some light on this...

Best regards,
Benoit

closed 2019-12-18 by bblanchon