bblanchon/ArduinoJson #1134: ArduinoJson 5: Error when parsing string from PubSubClient

albydnc, 2019-11-19:
Hi,

I have parse a big String obj coming from PubSubClient callback. 
I passed it to the parseObject method but I can't get it to parse. 
The string is a JSON object and I verified it was read right by printing it on Serial.
The same string defined in the code is parsed without any problem. Even when read from SD is okay.
Here the string
[CONFIG.TXT](https://github.com/bblanchon/ArduinoJson/files/3865359/CONFIG.TXT)
Here the callback method
```
void mqttCallback(char* topic, byte* payload, unsigned int len) {
#ifdef DEBUG
  SerialUSB.print("Message arrived [");
  SerialUSB.print(topic);
  SerialUSB.print("]: ");
  SerialUSB.write(payload, len);
  SerialUSB.println();
#endif
  String strPayload = String((char*)payload);
  if (String(topic) == mqttOtaTopic) {
    readJSONConfigFromString(strPayload);
    saveJSONConfig();
    readConfig();
  } else if (String(topic) == mqttCtrlTopic) {
    readJSONCommand(strPayload);
  }
}
```

Here the parsing method (a bit modified from your assistant)
```
void readJSONConfigFromString(String json) {
  JsonObject& root = jsonBuf.parseObject(json);
  JsonObject& global = root["global"];
  deviceName = global["name"].as<String>(); // "KAON"
  pollTime = global["poll"]; // 1
  snoozeTime = global["snooze"]; // 300
  updTime = global["upd"]; // 2
  alarmOn = global["alarm"]["on"]; // 8
  alarmOff = global["alarm"]["off"]; // 18
  adcBit = global["adc"]; // 16

  JsonArray& digital = root["digital"];
  for (int i = 0; i < 6; i++) {
    JsonObject& obj = digital[i];
    Dnames[i] = obj["name"].as<String>(); // "channel1"
    obj["type"].as<String>() == "INPUT" ? Dtype[i] = INPUT : Dtype[i] = OUTPUT; // "INPUT"
    Dalarm[i] = obj["alarm"]; // false
    Dthres[i] = obj["thres"]; // 0
    Dpriority[i] = obj["priority"]; // 0
    Ddelay[i] = obj["delay"]; // 0
  }
  JsonObject& ch7 = root["counters"]["CH7"];
  Cnames[0] = ch7["name"].as<String>(); // "channel1"
  Cunits[0] = ch7["unit"].as<String>(); // "m"
  Cmult[0] = ch7["mult"]; //1.0
  vals[6] = ch7["off"]; //1.0
  Calarm[0] = ch7["alarm"]; // false
  Cthres[0] = ch7["thres"]; // 0
  Cpriority[0] = ch7["priority"]; // 0
  Chour[0] = ch7["time"]; // 0

  JsonObject& ch8 = root["counters"]["CH8"];
  Cnames[1] = ch8["name"].as<String>(); // "channel1"
  Cunits[1] = ch8["unit"].as<String>(); // "m"
  Cmult[1] = ch8["mult"]; //1.0
  vals[7] = ch8["off"]; //1.0
  Calarm[1] = ch8["alarm"]; // false
  Cthres[1] = ch8["thres"]; // 0
  Cpriority[1] = ch8["priority"]; // 0
  Chour[1] = ch8["time"]; // 0

  JsonArray& analogIn = root["analogInput"];
  for (int i = 0; i < 4; i++) {
    JsonObject& obj = analogIn[i];
    Anames[i] = obj["name"].as<String>(); // "ai1"
    Aunits[i]  = obj["unit"].as<String>(); // "m3"
    Amult[i] = obj["mult"]; // 1
    Aoffset[i] = obj["off"]; // 0
    obj["type"].as<String>() == "4-20 mA" ? Atype[i] = mA_p : Atype[i] = V10_p; // "mA"
    Aalarm[i]  = obj["alarm"]; // false
    Alow[i] = obj["low"]; // 0
    Ahigh[i] = obj["high"]; // 0
    Apriority[i] = obj["priority"]; // 0
    Adelay[i] = obj["delay"]; // 0
  }
  JsonObject& aOut = root["analogOutput"];
  JsonObject& obj = aOut["AO1"];
  AOnames[0] = obj["name"].as<String>(); // "ai1"
  obj["type"].as<String>() == "4-20 mA" ? AOtype[0] = mA_p : AOtype[0] = V10_p; // "m3"
  JsonObject& obj2 = aOut["AO2"];
  AOnames[1] = obj2["name"].as<String>(); // "ai1"
  obj2["type"].as<String>() == "4-20 mA" ? AOtype[1] = mA_p : AOtype[1] = V10_p; // "m3"

  JsonObject& comp1 = root["logic"]["comp1"];
  comp1_a = charToChannel(comp1["a"]); // "CH1"
  comp1_b = charToChannel(comp1["b"]); // "CH2"
  comp1_fun = comp1["fun"].as<String>().charAt(0); // "="
  comp1_out = charToChannel(comp1["out"]); // "CH7"

  JsonObject& comp2 = root["logic"]["comp2"];
  comp2_a = charToChannel(comp2["a"]); // "AI1"
  comp2_b = charToChannel(comp2["b"]); // "AI2"
  comp2_fun = comp2["fun"].as<String>().charAt(0); // "<"
  comp2_out = charToChannel(comp2["out"]); // "CH8"

  JsonObject& pid = root["pid"];
  pidInput = charToChannel(pid["in"]); // "AI1"
  pidSet = pid["set"]; // 11
  pidMin = pid["min"]; // 0
  pidMax = pid["max"]; // 100
  pidP = pid["p"]; // 0.1
  pidI = pid["i"]; // 0
  pidD = pid["d"]; // 0.001
  pid["out"] == "AO1" ? pidOutput = 0 : pidOutput = 1; // "AO1"

  JsonObject& sms = root["sms"];
  smsAlive = sms["alive"]; // false
  smsTele = sms["tele"]; // false
  smsAlarm = sms["alarm"]; // true
  JsonArray& sms_target = sms["target"];
  nSms = sms_target.size();
  for (int i = 0; i < nSms; i++) smsTargets[i] = sms_target[i].as<String>();
  JsonObject& network = root["network"];
  JsonObject& network_apn = network["apn"];
  apnName = network_apn["name"].as<String>(); // "TM"
  apnUser = network_apn["user"].as<String>();
  apnPwd = network_apn["pw"].as<String>();

  JsonObject& network_mqtt = network["mqtt"];
  mqttUser = network_mqtt["user"].as<String>();
  mqttPwd = network_mqtt["pw"].as<String>();
  #ifdef DEBUG
    network_mqtt.printTo(SerialUSB);
    #endif
  mqttBroker = network_mqtt["broker"].as<String>(); // "mqtt.eclipse.org"
  mqttPort = network_mqtt["port"]; // 1883
  mqttId = network_mqtt["id"].as<String>(); // "dev1234"
  mqttAlive = network_mqtt["alive"]; // false
  mqttTele = network_mqtt["tele"]; // false
  mqttAlarm = network_mqtt["alarm"]; // true
  mqttCtrl = network_mqtt["control"]; // false
  mqttOta = network_mqtt["ota"]; // true

  JsonObject& network_rest = network["rest"];
  restUrl = network_rest["url"].as<String>();
  restPort = network_rest["port"]; // 8
  restId = network_rest["id"].as<String>(); // "dev1234"
  restAlive = network_rest["alive"]; // false
  restTele = network_rest["tele"]; // false
  restAlarm = network_rest["alarm"]; // true
  restCtrl = network_rest["control"]; // false
  restOta = network_rest["ota"]; // true
  jsonBuf.clear();
}

bblanchon, 2019-11-20:
Hi @albydnc,

The program duplicates the payload twice, so it's probably running out of memory.

You should pass directly the payload pointer to [`parseObject()`](https://arduinojson.org/v5/api/jsonbuffer/parseobject/).
Unfortunately, ArduinoJson 5 assumes that every string is null-terminated, so you cannot pass the `len` value to `parseObject()`.

I recommend that you [upgrade to ArduinoJson 6](https://arduinojson.org/v6/doc/upgrade/) because [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) returns an error code that tells you why parsing failed and supports an extra parameter to limit the length of the input.

Best Regards,
Benoit

albydnc, 2019-12-14:
Sorry for the late response.
I've upgraded the device to arduinoJson 6 as you suggested, I've shrunk the JsonDocuments but I can't get it to work. deserializeJson() gives me NoMemory error. How can I solve it?

bblanchon, 2019-12-16:
[NoMemory](https://arduinojson.org/v6/api/misc/deserializationerror/) means that the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) is too small, so you must increase its capacity.

Use the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) to compute the right capacity for your project.

Note: unlike `DynamicJsonBuffer`, `DynaicJsonDocument` doesn't extend when full, so you must set the right capacity.

albydnc, 2019-12-17:
I have used the assistant and then rounded up to the next thousand to be sure. I've tried increasing its capacity until I've doubled it, NoMemory is still given.

bblanchon, 2019-12-17:
In that case, it means that the [DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/) wasn't able to allocate the memory pool in the heap. You can confirm this by looking at [JsonDocument::capacity()](https://arduinojson.org/v6/api/jsondocument/capacity/) ; if it's 0, then it means the allocation failed.

To fix this problem, you must free some memory in the heap, or reduce the [fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html).

ewaldc, 2019-12-23:
@albydnc ,
Can you measure the free memory on your system just before parsing the string e.g. with [this library](https://github.com/mpflaga/Arduino-MemoryFree) ?  Your string should require a JsonDocument of only 4k when parsed directly with:
`DeserializationError deserializeJson(JsonDocument& doc, const char* payload, size_t len);`
Don't use the String library as it makes a full copy of your source char * string (payload) and then adds another few bytes for metadata (e.g. save some more memory by not using String(topic)).
There may even be another copy when passing the String to readJSONConfigFromString        
Ewald

albydnc, 2020-01-07:
@ewaldc Sorry for the delay, I have translated string to char *, but this did not help.
```
CONFIG INIT
ram free: 15443
SD init complete
FOUND JSON FILE
deserializeJson() status: NoMemory
ram free: 11267
```

This is the output

albydnc, 2020-01-07:
It seems that the error is coming from a bad serialization. By saving directly the string to the SD there is no issue, while serializeJson(doc, file) creates a file that gives the error above

ewaldc, 2020-01-08:
@albydnc,
As Benoit said, the _NoMemory_ error indicates lack of RAM, and e.g. not a syntax error.
You have only 15K of free memory and the JSON string already requires 8K for the _JsonDocument_ (+ whatever ArduinoJson requires as temporary storage during deserialization). Add to this all structures and buffers required by file system and file read buffers (worst case another scenario: another 8k), and it's likely running out of memory.  Even if you technically don't run out of memory, you may suffer heap fragmentation and thus there might not been a big enough free fragment available.  
Hence, IMHO, your best chance is to store the string in flash as a "_const char *_".  ArduinoJson will duplicate all string data, but that is OK as in total you will not need more than 8k + parsing overhead.

bblanchon, 2020-01-28:
@albydnc, did you make some progress?

albydnc, 2020-01-28:
@bblanchon I am following @ewaldc 's advice and I am still debugging the const char * implementation
I have got no success with the String class, my workaround is to save it to a file on a SD and then reboot the device.

albydnc, 2020-02-06:
I've configured  char arrays in stack to hold the values from and to json, but the issue I am facing now is to serialize a json from char arrays. the strcpy is not working and neither the = operator; resulting in a processor halt. I don't have at this moment the possibility to use a runtime debugger.\
How can I serialize those arrays?

bblanchon, 2020-02-07:
@albydnc, I don't understand what you mean by "serialize a json from char arrays", can you post some (pseudo-) code?

closed 2020-02-25 by bblanchon