bblanchon/ArduinoJson #1135: fatal error: ArduinoJson/Serialization/DynamicStringWriter.hpp: No such file or directory

afulloa, 2019-11-25:
Hi, 
when I compile de code of de example "JsonServer.ino" from de carpet "/Arduino/libraries/ArduinoJson/examples/JsonServer" y have de following error:
Plattform : linux ubuntu 16.04 AMD Athlon(tm) II X3 450 Processor × 3  with 121MB
IDE: Arduino 1.8.10
Compiler : AVRISK mkII

Error:->>>>
Se encontraron varias bibliotecas para "ArduinoJson.h"
Usado: /home/antonio/Arduino/libraries/ArduinoJson
In file included from /home/antonio/Arduino/libraries/ArduinoJson/src/ArduinoJson/Operators/VariantOr.hpp:9:0,
                 from /home/antonio/Arduino/libraries/ArduinoJson/src/ArduinoJson/Operators/VariantOperators.hpp:9,
                 from /home/antonio/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:12,
                 from /home/antonio/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from /home/antonio/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from /home/antonio/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:17,
                 from /home/antonio/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /home/antonio/Arduino/libraries/ArduinoJson/examples/JsonServer/JsonServer.ino:18:
/home/antonio/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:7:61: fatal error: ArduinoJson/Serialization/DynamicStringWriter.hpp: No such file or directory
compilation terminated.
exit status 1
Error compilando para la tarjeta NodeMCU-32S.

bblanchon, 2019-11-26:
Hi @afulloa,

The last time such an error occurred, it was because the library failed to upgrade.
Can you try to uninstall and reinstall ArduinoJson?

Best Regards,
Benoit

afulloa, 2019-11-28:
Hi @bblanchon, 
Resolved. I uninstall and reinstall and work fine.  
Thanks.

bblanchon, 2019-11-29:
Cool. Thanks for the update @afulloa.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project 😉

closed 2019-11-29 by bblanchon