bblanchon/ArduinoJson #1141: JsonObject value becomes null after changing it

HilalAH, 2019-12-08:
Every 3 seconds, i'm sending the below JSON where i change the value for the key 'deviceValue'

> `{"GetValueResponse":{"deviceValue":"11","id":"asd"}}

 it works at the start then it starts showing null values as shown below

> 15:43:09.372 -> {"GetValueResponse":{"value":"28","id":"asd"}} 15:43:19.512 ->
> 
> {"GetValueResponse":{"deviceValue":"20","id":"asd"}} 15:43:29.629 ->
> 
> {"GetValueResponse":{"deviceValue":"24","id":"asd"}} 15:43:39.738 ->
> 
> {"GetValueResponse":{"deviceValue":"26","id":null}} 15:43:49.875 ->
> 
> {"GetValueResponse":{"deviceValue":null,"id":null}} and the rest is as above
> 
> {"GetValueResponse":{"deviceValue":null,"id":null}}

this is the code that i'm using

`StaticJsonDocument<400>  getValueJsonRequest;
JsonObject getObject = getValueJsonRequest.createNestedObject("GetValue");

void setup(){
    getObject["value"] = String(random(0, 30));
      getObject["id"] = id;
      getValueJsonRequest.add(getObject);

}

void loop(){
      getObject["value"] = String(random(0, 30));
      Serial.println(getValueJsonRequest.as<String>());
      delay(2000);
 }`

bblanchon, 2019-12-09:
Hi @HilalAH,

Please read:

* [I found a memory leak in the library!](https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/)

Best Regards,
Benoit

HilalAH, 2019-12-09:
it worked thanks @bblanchon for your help

closed 2019-12-09 by HilalAH

bblanchon, 2019-12-10:
You're welcome @HilalAH.
Thank you for using the library.
Don't forget to [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show you support ;-)