bblanchon/ArduinoJson #1142: serializeMsgPack for unsigned char (uint8_t)

Piruthivi, 2019-12-12:
Hi, Am new to json format data and c++ too. I found it amusing, it is very easy to build a json when started to json generator example. And it is simple to use. Now lets see what my doubt is.

By using ArduinoJson library i can serializeJson to a character array and to a uart port like Serial1 or SerialUSB.
Similarly same can be done with serializeMsgPack. Now the tricky part is i need to use unsigned char for serialize message pack.
I did saw a "Custom Writer" could be a solution, but didnt have a clue to use it. Does it need to be added in the library files or in the program(.ino) files i have been using.

If i call serializeMsgPack(doc,(uint8_t) data) getting following error

_In member function 'size_t CustomWriter::write(uint8_t)':

abputilitypacket:22:9: error: request for member 'append' in '((CustomWriter*)this)->CustomWriter::str', which is of non-class type 'uint8_t {aka unsigned char}'

     str.append(1, static_cast<char>(c));

         ^_
If i could solve this i can move forward in my project. And thanks in advance.

bblanchon, 2019-12-12:
Hi @Piruthivi,

`serializeMsgPack()` should support `uint8_t`, it's an omission from my part.
I'll fix this as soon as possible.

Meanwhile, you can cast the pointer to `char*`:

```c++
serializeMsgPack(doc, reinterpret_cast<char*>(data), sizeof(data));
```

Best Regards,
Benoit

Piruthivi, 2019-12-13:
Thanks for the reply.
So i tried by type casting it.

_unsigned char hex_data[100];
serializeMsgPack(doc,(char *) hex_data);_

compilation shows

_error: macro "min" passed 3 arguments, but takes just 2

     min(const _Tp& __a, const _Tp& __b, _Compare __comp)_

Don't know am doing it right but just sharing the info.
So it didn't worked for me.

bblanchon, 2019-12-13:
Hi @Piruthivi,

I just pushed a fix for this issue. Please try with the "6.x" branch; you should be able to write `serializeJson(doc, hex_data)` without casting.

Best Regards,
Benoit

commit 8bf6f6e0 (2019-12-13): Added support `uint8_t` for `serializeMsgPack()` (closes #1142)
closed 2019-12-13 by bblanchon