bblanchon/ArduinoJson #1143: JSON buffer size from function call help

salvq, 2019-12-12:
Currently I am using ArduinoJson 6.13.0 and it just works as expected 👍

It is rather question / help than issue...

I am using loop sketch to post mqtt message to broker, for this purpose I need size of msg due to QoS1 reason.

For some other purposes like logging data to SD I do not need buffer size so I created also separate function to be called anytime I need to get json data from digital inputs.

The issue I am facing just now is how to get rid of main sketch routine and call just the routine before mqtt section. However the issue is the buffer size. 

**How to correctly make function routine to get json message and buffer size into mqtt routine and get rid of main sketch routine ?**

**Compile OK, wrong size calculated (size_t 152, sizeof 256)**
```
      mqttClient.beginMessage(outTopic.c_str(), sizeof(readDI2Json()), retained, qos, dup);
      mqttClient.print(readDI2Json);
      mqttClient.endMessage();
```
**Compile NOK, string to size_t issue**
```
      mqttClient.beginMessage(outTopic.c_str(), size_t(readDI2Json()), retained, qos, dup);
      mqttClient.print(readDI2Json());
      mqttClient.endMessage();
```

**Working main sketch:**
```
      StaticJsonDocument<300> doc;  /* Reserve memory space for the Json buffer */

      doc["dateStamp"] = getDateStamp();
      doc["timeStamp"] = getTimeStamp();
      doc["deviceId"] = deviceId;
      doc["messageId"] = messageCount++;
      doc["yellow"] = !digitalRead(0);
      doc["green"] = !digitalRead(1);
      doc["cycle"] = !digitalRead(15);
      doc["red"] = !digitalRead(16);
      doc["reject"] = !digitalRead(17);

      char buffer[256];  /* character string containing the JSON payload */
      serializeJson(doc, buffer);  /* convert to buffer, which is the string to be used as the message */
  
      mqttClient.beginMessage(outTopic.c_str(), size_t(serializeJson(doc, buffer)), retained, qos, dup);
      mqttClient.print(buffer);
      mqttClient.endMessage();
```

**Function call:**
```
String readDI2Json() {
  StaticJsonDocument<300> doc;  /* Reserve memory space for the Json buffer */

  doc["dateStamp"] = getDateStamp();
  doc["timeStamp"] = getTimeStamp();
  doc["deviceId"] = deviceId;
  doc["messageId"] = messageCount++;
  doc["yellow"] = !digitalRead(0);
  doc["green"] = !digitalRead(1);
  doc["cycle"] = !digitalRead(15);
  doc["red"] = !digitalRead(16);
  doc["reject"] = !digitalRead(17);

  char buffer[256];  /* character string containing the JSON payload */
  serializeJson(doc, buffer);  /* convert to buffer, which is the string to be used as the message */

  return buffer;
}
```

bblanchon, 2019-12-13:
Hi @salvq,

In C++, the `sizeof()` operator is evaluated at **compile-time** and returns the **size of the type**.
For example, `sizeof(readDI2Json())` is equilvalent to `sizeof(String)` and returns the size of the `String` type (note that the `readDI2Json()` function is not evaluated here).

In this case, the best solution is to use the result of [`serializeJson()`](https://arduinojson.org/v6/api/json/serializejson/); it contains the length of the serialized document.

Here is how you can use it in the main loop:

```c++
// ... same as before
char buffer[256];
size_t n = serializeJson(doc, buffer);
  
mqttClient.beginMessage(outTopic, n, retained, qos, dup);
mqttClient.print(buffer);
mqttClient.endMessage();
```

Now, if you want to extract a function to generate the JSON document, I recommend that you stay away from `String` and use a `char[]` instead:

```c+
size_t readDI2Json(char* buffer, size_t buffer_size) {
  // ...same as before...
  return serializeJson(doc, buffer, buffer_size);
}

// then, in main loop:
char buffer[256];
size_t n = readDI2Json(doc, buffer, sizeof(buffer));
  
mqttClient.beginMessage(outTopic, n, retained, qos, dup);
mqttClient.print(buffer);
mqttClient.endMessage();
```

Best Regards,
Benoit

bblanchon, 2019-12-13:
PS: I added [How to use ArduinoJson with ArduinoMqttClient](https://arduinojson.org/v6/faq/how-to-use-arduinojson-with-arduinomqttclient/) in the FAQ.

salvq, 2019-12-14:
@bblanchon thanks for help, your approach works just fine, just small adjustment...
intead of `size_t n = readDI2Json(doc, buffer, sizeof(buffer));` need to use `size_t n = readDI2Json(buffer, sizeof(buffer));`

Final sketch looks like this:
**Function**
```
size_t readDI2Json(char* buffer, size_t buffer_size) {
  
  StaticJsonDocument<300> doc;  // Reserve memory space for the Json buffer
  
  doc["dateStamp"] = getDateStamp();
  doc["timeStamp"] = getTimeStamp();
  doc["deviceId"] = deviceId;
  doc["messageId"] = ++messageCount;
  doc["yellow"] = !digitalRead(0);
  doc["green"] = !digitalRead(1);
  doc["cycle"] = !digitalRead(15);
  doc["red"] = !digitalRead(16);
  doc["reject"] = !digitalRead(17);

  return serializeJson(doc, buffer, buffer_size);
}
```
**Main loop**
```
      char buffer[256];

      mqttClient.beginMessage(outTopic.c_str(), size_t(readDI2Json(buffer, sizeof(buffer))), retained, qos, dup);
      mqttClient.print(buffer);
      mqttClient.endMessage();
```

bblanchon, 2019-12-16:
Excellent!
That's precisely how one should use ArduinoJson: a `JsonDocument` scoped to the serialization function.

closed 2019-12-16 by bblanchon